labgrid-project/labgrid #1086: labgrid-suggest doesn't notice new fastboot devices

khilman, 2023-02-01:
labgrid-suggest is a great tool when adding new devices, and it works great for USB serial ports etc.  But it doesn't detect new USB fastboot devices.  

Please consider it a feature request to add detection/report of fastboot devices to labgrid-suggest.  At least ones that are using the standard USB vendor/device IDs from google for this.

jluebbe, 2023-02-01:
Devices with vendor 1D6B and model 0104 should work already. Could you post the output of `udevadm monitor -p` when connecting the device that is not detected?

khilman, 2023-02-01:
This one has a different vendor ID:
```
$ lsusb |grep -i google
Bus 001 Device 068: ID 18d1:fada Google Inc. USB download gadget
```

Also, here is the `udevadm monitor -p ` output when the device is plugged in:

[fastboot.log](https://github.com/labgrid-project/labgrid/files/10559750/fastboot.log)

khilman, 2023-02-02:
I just noticed `adb_user=yes` in the monitor output.  Perhaps combining this with `ID_VENDOR_FROM_DATABASE=Google Inc.` would be enough info to decide this is an adb/fastboot device instead of relying on keeping a list of diffferent `ID_MODEL_ID` ?

Emantor, 2023-02-07:
This will work with distros that use  [android-udev-rules](https://github.com/M0Rf30/android-udev-rules), which includes at least NixOS and Arch Linux, but not debian (or I am bad at searching).

Emantor, 2023-02-07:
The problem with this is that we not only identify fastboot devices, but also devices which only have `adb` enabled.

jluebbe, 2023-02-07:
As long as we don't have a ADBDriver, matching on adb_user=yes is probably fine.

When we add ADB support, we'll need to distinguish between them somehow, as otherwise both resources will become available at the same time, which is makes it harder to synchronize with target state changes.

PR #1089 (2023-02-07): resource/udev: allow fastboot with adb_user=yes
**Description**
Distros which implement the android-udev-rules from github use the adb_user property to indicate that the device is valid for fastboot. Make use of this property for labgrid which removes the need for more hardcoded USB IDs.

**Checklist**
- [x] PR has been tested

Fixes #1086
closed 2023-02-07 by Emantor