labgrid-project/labgrid #1121: labgrid-client power fails in CI runner

kjeldflarup, 2023-03-30:
I'm running labgrid from Gitlab CI, which works fine, but now I have added a power off in the end of the script, and now I see this weird behaviour

```
$ labgrid-client -p $LG_PLACE power off
labgrid-client: error: multiple resources matching <class 'labgrid.resource.networkservice.NetworkService'> found in Target(name='main', env=Environment(config_file='/etc/labgrid.yaml'))
Selected role main from configuration file
Found multiple resources but no name was given, available names:
rpi
ssh
```

Why does the power command fiddle with the network service?  
Can it be something in the environment that is different?

Bastian-Krause, 2023-03-30:
Please share the output of `labgrid-client -p $LG_PLACE show`.

Bastian-Krause, 2023-03-30:
Seems related to #1049, so probably nothing to worry about regarding #1111 and #1112.

kjeldflarup, 2023-03-30:
Thanks for the tip.
```
LG_ENV= labgrid-client power off
```
Solves the problem. Thanks for the tip.

But this does not seem to be affected
```
labgrid-client release
```

closed 2023-03-30 by kjeldflarup

Bastian-Krause, 2023-03-30:
> But this does not seem to be affected
> 
> ```
> labgrid-client release
> ```

True, the resource and driver instantiation happens on target retrieval:
 https://github.com/labgrid-project/labgrid/blob/88dc6a7d78298a6f174be87594d37979cf8bd5d9/labgrid/remote/client.py#L660

This again is only done for sub commands that interact with the target. `acquire`/`release` don't do that.