labgrid-project/labgrid #1144: Declaring features in exporter

kjeldflarup, 2023-04-24:
I can declare features in the .yaml file https://labgrid.readthedocs.io/en/latest/usage.html#feature-flags  
However that goes for all targets, and if I have targets with ex. hardware differences, then I would like to specify that in the exporter, so @pytest.mark.lg_feature would enable or disable a certain test. 

However it does not seem possible to do this. 

The issue is similar to https://github.com/labgrid-project/labgrid/issues/1057, where the workaound was to set a tag in the coordinater which I could read. However I cannot pass this to pytest.mark.

Emantor, 2023-04-24:
> I can declare features in the .yaml file https://labgrid.readthedocs.io/en/latest/usage.html#feature-flags However that goes for all targets, and if I have targets with ex. hardware differences, [...]

Feature flags support both global and target features as documented in the section you linked to.
Target feature flags are documented first, global flags are introduced at the end of the section.

kjeldflarup, 2023-04-24:
As far as I can see, that section does not cover the exporter.

Emantor, 2023-04-24:
That is correct, since feature flags are purely a configuration yaml feature. Matching the correct yaml configuration to the correct remote place to be used for test execution is currently left to the CI pipeline. This means that for the following places:
```
Place: a
Features: camera, display

Place: b
Features: display
```
It is required that there are two different yaml files using the same python testfiles but defining different features.
This also matches the `RemotePlace`, since the place name needs to be passed explicitly.

What you probably want is new functionality that is able to turn tags into features. However since features can be anything at the moment and are not retrievable inside of labgrid, we would need to impose restrictions on the possible names that can be used for tags to carve out a space for features. I am not sure we want to do that, given that most CI pipelines have different requirements how information for different boards is passed to the testsuite, so my preference is for this mechanism to remain out of band for labgrid. However I am open to extending labgrid with the ability to retrieve tags for `RemotePlaces`, probably via a property on the `RemotePlace` instance.

Bastian-Krause, 2023-07-23:
I think all use cases are now covered, so I'm closing this. Please speak up if that's not correct.

closed 2023-07-23 by Bastian-Krause