labgrid-project/labgrid #1173: Resource names must be unique per resource class, not globally

Bastian-Krause, 2023-05-08:
This would allow multiple resources named "default" (which was intended by #1112).

Pointed out by @nlabriet in https://github.com/labgrid-project/labgrid/issues/1163#issuecomment-1534341129:
> I changed my configuration to remove names from the exporter and set them in the coordinator with `add-named-match`. It works, except when I need to have 2 (or more) resources with the same name ("default").
> 
> [`get_target_resources()`](https://github.com/labgrid-project/labgrid/blob/e1585887fde9111e6672a4df4c616fa5f4b54963/labgrid/remote/client.py#L628) creates a dictionary with the name as the key. In my case, the name should be "default", but the resource classes are different.
> 
> I tested using the (name, class) tuple as key and it solves my issue. Does this seem reasonable to you?

Emantor, 2023-05-09:
From a quick look this only seems to be broken in `labgrid-client` but not in the library itself. A test for this should be straightforward to implement, I'll take a look within the next maintainer slot if nobody get to it beforehand.

Bastian-Krause, 2023-05-09:
> From a quick look this only seems to be broken in `labgrid-client` but not in the library itself.

I don't think so. It's correct that the code that needs fixing is located in `labgrid/remote/client.py`, but `get_target_resources()` is used in `labgrid/resource/remote.py` for the RemotePlace handling. So the library is also affected by this bug.

Emantor, 2023-05-09:
> I don't think so. It's correct that the code that needs fixing is located in `labgrid/remote/client.py`, but `get_target_resources()` is used in `labgrid/resource/remote.py` for the RemotePlace handling. So the library is also affected by this bug.

Ah, right. `get_target_resources()` and its users need to be fixed.

PR #1176 (2023-05-17): Unique resource name per resource class
**Description**
Allow using the same resource name across classes.
I tested it in my setup and created a test that fails without this fix.

**Checklist**
- [ ] Documentation for the feature
  - No documentation on this fix
- [X] Tests for the feature 
- [X] PR has been tested

Fixes #1173
commit 1f0c9042 (2023-05-17): Unique resource name per resource class

Fixes #1173
closed 2023-05-17 by jluebbe