labgrid-project/labgrid #1185: Couldn't start crossbar using default "config-anonymous.yaml"

Sriram-JI, 2023-05-19:
Hi,
    I have been trying to run crossbar with default configuration file  - config-anonymous.yaml.
    
    `2023-05-19T17:04:12+0530 [Controller 369497] Ok, local node configuration ran successfully.
2023-05-19T17:04:12+0530 [Controller 369497] <crossbar.node.node.Node.boot>::NODE_BOOT_COMPLETE
2023-05-19T17:04:12+0530 [Guest      369511] Traceback (most recent call last):
2023-05-19T17:04:12+0530 [Guest      369511] File "<frozen runpy>", line 198, in _run_module_as_main
2023-05-19T17:04:12+0530 [Guest      369511] File "<frozen runpy>", line 88, in _run_code
2023-05-19T17:04:12+0530 [Guest      369511] File "/home/sriram/Documents/Tasks/8_CRATE/labgrid/crossbar-venv/lib/python3.11/site-packages/labgrid/remote/coordinator.py", line 14, in <module>
2023-05-19T17:04:12+0530 [Guest      369511] from autobahn.asyncio.wamp import ApplicationRunner, ApplicationSession
2023-05-19T17:04:12+0530 [Guest      369511] File "/home/sriram/Documents/Tasks/8_CRATE/labgrid/crossbar-venv/lib/python3.11/site-packages/autobahn/asyncio/__init__.py", line 32, in <module>
2023-05-19T17:04:12+0530 [Guest      369511] from autobahn.asyncio.websocket import \
2023-05-19T17:04:12+0530 [Guest      369511] File "/home/sriram/Documents/Tasks/8_CRATE/labgrid/crossbar-venv/lib/python3.11/site-packages/autobahn/asyncio/websocket.py", line 36, in <module>
2023-05-19T17:04:12+0530 [Guest      369511] from autobahn.wamp import websocket
2023-05-19T17:04:12+0530 [Guest      369511] File "/home/sriram/Documents/Tasks/8_CRATE/labgrid/crossbar-venv/lib/python3.11/site-packages/autobahn/wamp/websocket.py", line 32, in <module>
2023-05-19T17:04:12+0530 [Guest      369511] from autobahn.websocket import protocol
2023-05-19T17:04:12+0530 [Guest      369511] File "/home/sriram/Documents/Tasks/8_CRATE/labgrid/crossbar-venv/lib/python3.11/site-packages/autobahn/websocket/protocol.py", line 2533, in <module>
2023-05-19T17:04:12+0530 [Guest      369511] class WebSocketServerProtocol(WebSocketProtocol):
2023-05-19T17:04:12+0530 [Guest      369511] File "/home/sriram/Documents/Tasks/8_CRATE/labgrid/crossbar-venv/lib/python3.11/site-packages/autobahn/websocket/protocol.py", line 2538, in WebSocketServerProtocol
2023-05-19T17:04:12+0530 [Guest      369511] log = txaio.make_logger()
2023-05-19T17:04:12+0530 [Guest      369511] ^^^^^^^
2023-05-19T17:04:12+0530 [Guest      369511] ^^^^^^^^^^^^
2023-05-19T17:04:12+0530 [Guest      369511] File "/home/sriram/Documents/Tasks/8_CRATE/labgrid/crossbar-venv/lib/python3.11/site-packages/txaio/_unframework.py", line 41, in _throw_usage_error
2023-05-19T17:04:12+0530 [Guest      369511] raise RuntimeError(
2023-05-19T17:04:12+0530 [Guest      369511] RuntimeError: To use txaio, you must first select a framework with .use_twisted() or .use_asyncio()
2023-05-19T17:04:12+0530 [Controller 369497] Guest coordinator exited with error A process has ended with a probable error condition: process ended with exit code 1.
`

Not sure of this error.

Emantor, 2023-05-19:
What does the crossbar config look like?

Sriram-JI, 2023-05-19:
yes emantor i am in master branch.

config file:
version: 2
workers:
- type: router
  realms:
  - name: realm1
    roles:
    - name: public
      permissions:
      - uri: ''
        match: prefix
        allow:
          call: true
          register: true
          publish: true
          subscribe: true
        disclose:
          caller: true
          publisher: true
        cache: true
  transports:
  - type: web
    endpoint:
      type: tcp
      port: 20408
    paths:
      /:
        type: static
        directory: ../web
      ws:
        type: websocket
        auth:
          anonymous:
            type: static
            role: public
- id: coordinator
  type: guest
  executable: python3
  arguments:
  - -m
  - labgrid.remote.coordinator
  options:
    workdir: .
    env:
      vars:
        WS: ws://localhost:20408/ws

Sriram-JI, 2023-05-19:
```
ansicolors==1.1.8
argon2-cffi==21.3.0
argon2-cffi-bindings==21.2.0
attrs==23.1.0
autobahn==22.4.1
Automat==22.10.0
bcrypt==4.0.1
bitstring==4.0.2
cbor==1.0.0
cbor2==5.4.6
certifi==2023.5.7
cffi==1.15.1
charset-normalizer==3.1.0
click==8.1.3
colorama==0.4.6
constantly==15.1.0
crossbar==21.3.1
cryptography==40.0.2
flatbuffers==23.5.9
h2==4.1.0
hpack==4.0.0
hyperframe==6.0.1
hyperlink==21.0.0
idna==2.5
importlib-resources==5.12.0
incremental==22.10.0
iniconfig==2.0.0
Jinja2==3.1.2
labgrid @ file:///home/sriram/Documents/Tasks/8_CRATE/labgrid
lmdb==1.4.1
MarkupSafe==2.1.2
mistune==2.0.5
msgpack==1.0.5
netaddr==0.8.0
numpy==1.24.3
packaging==23.1
passlib==1.7.4
pexpect==4.8.0
pluggy==1.0.0
priority==2.0.0
psutil==5.9.5
ptyprocess==0.7.0
py-ubjson==0.16.1
pyasn1==0.5.0
pyasn1-modules==0.3.0
pycparser==2.21
Pygments==2.15.1
PyNaCl==1.5.0
pyOpenSSL==23.1.1
PyQRCode==1.2.1
pyserial-labgrid==3.5.0.2
pytest==7.3.1
python-snappy==0.6.1
PyTrie==0.4.0
pyudev==0.24.1
pyusb==1.2.1
PyYAML==6.0
requests==2.30.0
sdnotify==0.3.2
service-identity==21.1.0
setproctitle==1.3.2
six==1.16.0
sortedcontainers==2.4.0
treq==22.2.0
Twisted==22.10.0
txaio==23.1.1
txtorcon==23.0.0
typing_extensions==4.5.0
u-msgpack-python==2.8.0
ujson==5.7.0
urllib3==1.24.3
watchdog==3.0.0
Werkzeug==2.3.4
xmodem==0.4.6
zlmdb==23.1.1
zope.interface==6.0
```

Bastian-Krause, 2023-05-19:
It would also be helpful to see how you execute crossbar.

Note that the executable specified in the crossbar config needs to point to a python interpreter inside the labgrid venv (not the crossbar venv). Your setup only works if you activated the labgrid venv and call crossbar from outside its venv.

The recommended way is documented [here](https://labgrid.readthedocs.io/en/latest/getting_started.html#coordinator).

Bastian-Krause, 2023-05-19:
You've installed crossbar and labgrid into the same virtualenv. That's neither recommended nor tested. See the link above for the recommended setup. Maybe we need to document the reasoning behind this better..

Sriram-JI, 2023-05-19:
Ok thanks @Bastian-Krause for details.
Its working now will further explore labgrid and its features...

closed 2023-05-19 by Sriram-JI