labgrid-project/labgrid #1199: ssh error: <name> too long for Unix domain socket

cruise-bryan-murdock, 2023-06-06:
My corporate network has some long usernames and hostnames and the default settings for ssh use a `ControlPath` that includes those long names.  This can result in socket names that are too long which gives the error in the title of this issue.  [I found the solution to this](https://gist.github.com/andyvanee/bcf95b1044b80e72b4a42933549a079b) which is to set `ControlPath=~/.ssh/master-socket/%C` but changing my `.ssh/config` file does not change the behavior of labgrid.  Somewhere in the labgrid code, I guess, the `ControlPath` needs to be changed to use `%C`

Bastian-Krause, 2023-06-07:
We had #949 before. Seems like hostnames can consist of up to 253 characters while Unix domain sockets are limited to 108 characters (or 104 on MacOS). labgrid constructs custom ControlPaths. I guess hashing the host name should be okay.

@Emantor You implemented the SSH connections, what do you think?

Emantor, 2023-06-07:
Yep, sounds like a sound idea.

cruise-bryan-murdock, 2023-06-07:
I tried hashing today and that didn't help.  The real problem is the length of the path that tempfile.mkdtemp returns :-(

Trying to figure out if there's a way to shorten that up

cruise-bryan-murdock, 2023-06-08:
I don't think you can tell tempfile to make a shorter path.  [A suggestion I found elsewhere](https://github.com/python/cpython/issues/93852#issuecomment-1156361215) was to just put the socket in the cwd.  I tried that and it worked for me.  Here's the diff:

```
diff --git a/labgrid/util/ssh.py b/labgrid/util/ssh.py
index 0f51bbc..ee648cc 100644
--- a/labgrid/util/ssh.py
+++ b/labgrid/util/ssh.py
@@ -134,7 +134,7 @@ class SSHConnection:
         default=False, init=False, validator=attr.validators.instance_of(bool)
     )
     _tmpdir = attr.ib(
-        default=attr.Factory(lambda: tempfile.mkdtemp(prefix="lg-con-")),
+        default=attr.Factory(lambda: tempfile.mkdtemp(prefix="lg-con-", dir=os.getcwd())),
         init=False,
         validator=attr.validators.instance_of(str)
     )
```

Bastian-Krause, 2023-06-08:
What does `import tempfile; print(tempfile.gettempdir())` print at your end?

cruise-bryan-murdock, 2023-06-08:
On an Ubuntu 20.04 machine it returns
```
/tmp
```

On my macbook pro it returns:
```
/var/folders/f3/tpfpr6gs47gdkn_9jxjb45r80000gq/T
```
which is 48 characters.  A typical fully qualified hostname for one of our exporter machines is 29 characters, for a total of 77 characters.  The full socket path comes out looking something like this:

```
/var/folders/f3/tpfpr6gs47gdkn_9jxjb45r80000gq/T/lg-con-8zznjgtb/control-<29-character-hostname>.mHlAjRwVyLDanIyq
```
which comes out to 119 characters.

Bastian-Krause, 2023-06-21:
> On an Ubuntu 20.04 machine it returns
> 
> ```
> /tmp
> ```
> 
> On my macbook pro it returns:
> 
> ```
> /var/folders/f3/tpfpr6gs47gdkn_9jxjb45r80000gq/T
> ```
> 
> which is 48 characters. A typical fully qualified hostname for one of our exporter machines is 29 characters, for a total of 48+29 = 77 characters. The full socket path, however, comes out even longer, looking something like this:
> 
> ```
> /var/folders/f3/tpfpr6gs47gdkn_9jxjb45r80000gq/T/lg-con-8zznjgtb/control-<29-character-hostname>.mHlAjRwVyLDanIyq
> ```

No idea where `.mHlAjRwVyLDanIyq` is coming from.

> 
> that's 119 characters.

I guess there's not a lot we can do about this, apart from using a different parent directory, but creating control sockets in unexpected locations (that work on both platforms) is not really a nice solution.

I'm a little hesitant. We've never advertised MacOS support, but we merged some PRs with minor fixes for MacOS. @jluebbe @Emantor What do you think about this?

cruise-bryan-murdock, 2023-06-21:
That suffix on the hostname seems to just be another quick of macOS.

Other people have this same "path too long" problem on macOS and they seem to solve it by simply hard-coding things to use `/tmp` on macOS 🤷 

https://search.brave.com/search?q=macos+%22too+long+for+Unix+domain+socket%22

Specifically:  https://docs.tarmak.io/known-issues/path-too-long.html

cruise-bryan-murdock, 2023-07-03:
We gave up on MacOS and switched to Linux.  This seems like a general MacOS problem.  Whoever came up with that temp file path did not consider the limitation on unix domain socket path length.

closed 2023-07-03 by cruise-bryan-murdock