labgrid-project/labgrid #1209: pytest plugin should not be registered by default

edersondisouza, 2023-06-14:
Why is it that the pytest plugin is registered by default, and not only on the "dev environment"?

This can cause issues if someone is running an unrelated pytest session on a different project, and labgrid plugin is loaded by pytest but fails (due for instance, the other project setting `-p no:logging`).

Is there a good reason for that? It seems to me that running pytest is more important for development anyway.

Emantor, 2023-06-14:
Pytest always loads all plugins by default, see [Pytest plugin docs](https://docs.pytest.org/en/7.1.x/how-to/plugins.html#deactivating-unregistering-a-plugin-by-name). If labgrid is available within your environment, pytest will load the labgrid pytest plugin for testing.

However the labgrid pytest plugin failing if the logging plugin is no loaded is a bug. We should fix that and maybe fallback to simple printing in this case.

edersondisouza, 2023-06-14:
I see - then I guess fixing labgrid plugin is the way to go. Should I file a specific issue for that?

Emantor, 2023-06-15:
> I see - then I guess fixing labgrid plugin is the way to go. Should I file a specific issue for that?

Not necessary, we can reuse this one and adjust the title.

PR #1210 (2023-06-16): Fix handling disabled pytest logging plugin
**Description**

Adjust our pytestplugin to handle the case where the logging plugin is disabled.

**Checklist**
- [x] PR has been tested

Fixes #1209
closed 2023-06-16 by jluebbe