labgrid-project/labgrid #1263: Gude power strip access failed

est-flouveau, 2023-09-04:
Hi Everyone,

I'm using labgrid in its release v23.0.3. 
I bought a EPC 1202 from Gude (firmware v1.11) and want to use this driver to control my device power cycle.

I try all gude driver without success.
Here is the output of the ov.html:
```html
<!DOCTYPE html>
<html><head><!--
<meta http-equiv="powerstate" content="Power Port,1">
<meta http-equiv="powerstate" content="Power Port,1">
<meta http-equiv="powerstate" content="Power Port,1">
<meta http-equiv="powerstate" content="Power Port,1">
-->
</head><body></body></html>
```

And this is not compatible with available power drivers.
- gude.py => driver does not search for ov.html
- gude24.py => pattern is not compatible with ov.hml
- gude8031.py => driver does not search for ov.html
- gude8225.py => driver does not search for ov.html
- gude8316.py => pattern is not compatible with ov.hml

I'm quite confused with this and don't know how to handle it.
I modified gude24.py to fit the ov.html output, but don't know if it's the correct way.

Moreover, we can use http://192.168.0.2/statusjsn.js?components=1 to get statuses of outputs and parse using json.

Emantor, 2023-09-06:
Is this by chance already supported inside of pdudaemon? In that case I'd recommend using pdudaemon instead.

est-flouveau, 2023-09-07:
It seems that the 1202 has been recently integrated inside pdudaemon using telnet:

https://github.com/pdudaemon/pdudaemon/commit/6739a7dfe1f0d84e97b4bb7fa1930841823909d2

I will give it a try. Thanks for the tips

est-flouveau, 2023-10-05:
I haven't had time to test this solution yet.