labgrid-project/labgrid #1269: Trying to set up USBMassStorage labgrid.exceptions.NoSupplierFoundError

JohnTordur, 2023-09-14:
I'm getting a labgrid.exceptions.NoSupplierFoundError.

I'm trying to set the usbsdmux up with both the mux and mass storage driver:
Here is what I've tried:
Added the remote user to the `disk` group. 

Here is the full error when trying out the strategy:
```
  DEBUG: Starting session with "ws://coordinator.lg.swtech.dk/ws", realm: "realm1"
  DEBUG: expanded remote resources for place jenis: [NetworkPowerPort(target=Target(name='jenis', env=None), name='NetworkPowerPort', state=<BindingState.bound: 1>, avail=True, model='shelly_gen1', host='http://192.168.0.2', index='0'), NetworkService(target=Target(name='jenis', env=None), name='NetworkService', state=<BindingState.bound: 1>, avail=True, address='192.168.0.75', username='root', password='', port=22), NetworkUSBMassStorage(target=Target(name='jenis', env=None), name='USBMassStorage', state=<BindingState.bound: 1>, avail=True, host='helios.swtech.dk', timeout=10.0, busnum=1, devnum=7, path='/dev/sda', vendor_id=1060, model_id=16449), RemoteNetworkInterface(target=Target(name='jenis', env=None), name='USBNetworkInterface', state=<BindingState.bound: 1>, avail=True, host='helios.swtech.dk', timeout=2.0, ifname='enx00e04c680429'), NetworkUSBSDMuxDevice(target=Target(name='jenis', env=None), name='USBSDMuxDevice', state=<BindingState.bound: 1>, avail=True, host='helios.swtech.dk', timeout=10.0, busnum=1, devnum=7, path='/dev/sda', vendor_id=1060, model_id=16449, control_path='/dev/sg0'), NetworkSerialPort(target=Target(name='jenis', env=None), name='USBSerialPort', state=<BindingState.bound: 1>, avail=True, host='helios.swtech.dk', port=39489, speed=115200, protocol='rfc2217')]
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/dist-packages/labgrid/remote/client.py", line 704, in _get_driver_or_new
    return target.get_driver(cls, name=name, activate=activate)
  File "/usr/local/lib/python3.10/dist-packages/labgrid/target.py", line 230, in get_driver
    return self._get_driver(cls, name=name, activate=activate)
  File "/usr/local/lib/python3.10/dist-packages/labgrid/target.py", line 183, in _get_driver
    raise NoDriverFoundError(
labgrid.exceptions.NoDriverFoundError: no <class 'labgrid.driver.usbstoragedriver.USBStorageDriver'> driver found in Target(name='jenis', env=None)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.10/dist-packages/labgrid/remote/client.py", line 1871, in main
    args.func(session)
  File "/usr/local/lib/python3.10/dist-packages/labgrid/remote/client.py", line 1209, in write_image
    drv = self._get_driver_or_new(target, "USBStorageDriver", activate=False, name=name)
  File "/usr/local/lib/python3.10/dist-packages/labgrid/remote/client.py", line 717, in _get_driver_or_new
    drv = cls(target, name=name)
  File "<attrs generated init labgrid.driver.usbstoragedriver.USBStorageDriver>", line 9, in __init__
    self.__attrs_post_init__()
  File "/usr/local/lib/python3.10/dist-packages/labgrid/driver/usbstoragedriver.py", line 46, in __attrs_post_init__
    super().__attrs_post_init__()
  File "/usr/local/lib/python3.10/dist-packages/labgrid/driver/common.py", line 24, in __attrs_post_init__
    super().__attrs_post_init__()
  File "/usr/local/lib/python3.10/dist-packages/labgrid/binding.py", line 55, in __attrs_post_init__
    target.bind(self)
  File "/usr/local/lib/python3.10/dist-packages/labgrid/target.py", line 423, in bind
    return self.bind_driver(bindable)
  File "/usr/local/lib/python3.10/dist-packages/labgrid/target.py", line 378, in bind_driver
    raise NoSupplierFoundError(f"conflicting suppliers matching {requirements} found in target {self}")  # pylint: disable=line-too-long
labgrid.exceptions.NoSupplierFoundError: conflicting suppliers matching {'NetworkUSBMassStorage', 'NetworkUSBSDMuxDevice', 'USBSDWireDevice', 'USBMassStorage', 'NetworkUSBSDWireDevice', 'USBSDMuxDevice'} found in target Target(name='jenis', env=None)
```

Here is my exporter config:
```
alpha-group:
  location: alpha
  USBSerialPort:
    match:
      ID_PATH: pci-0000:01:00.0-usb-0:1.3.2:1.0
    speed: 115200
  USBSDMuxDevice:
    match:
      ID_PATH: pci-0000:01:00.0-usb-0:1.4.1
  USBNetworkInterface:
    match:
      ID_PATH: pci-0000:01:00.0-usb-0:1.2:1.0
  NetworkPowerPort:
    model: shelly_gen1
    host: http://192.168.0.2
    index: 0
  NetworkService:
    address: 192.168.0.75
    username: root
  USBMassStorage:
    match:
      ID_PATH: pci-0000:01:00.0-usb-0:1.4.1:1.0-scsi-0:0:0:0
```

Here is my environment config:
[... 128 lines cut ...]
            self.target.activate(self.uboot)
        elif status == Status.shell:
            # transition to uboot
            self.transition(Status.uboot)
            self.uboot.boot("")
            self.uboot.await_boot()
            self.target.activate(self.shell)
            self.shell.run("systemctl is-system-running --wait")
        else:
            raise StrategyError(f"no transition found from {self.status} to {status}")
        self.status = status

    def force(self, status):
        if not isinstance(status, Status):
            status = Status[status]
        if status == Status.off:
            self.target.activate(self.power)
        elif status == Status.uboot:
            self.target.activate(self.uboot)
        elif status == Status.shell:
            self.target.activate(self.shell)
        else:
            raise StrategyError("can not force state {}".format(status))
        self.status = status
```

Emantor, 2023-09-14:
Please also post `labgrid-client -p jenis show`.

JohnTordur, 2023-09-14:
> Please also post `labgrid-client -p jenis show`.

Sure here:
```
Place 'jenis':
  matches:
    */alpha-group/*
  acquired: cfff0cf10bf5/vscode
  acquired resources:
    helios/alpha-group/NetworkPowerPort/NetworkPowerPort
    helios/alpha-group/NetworkSerialPort/USBSerialPort
    helios/alpha-group/NetworkService/NetworkService
    helios/alpha-group/NetworkUSBMassStorage/USBMassStorage
    helios/alpha-group/NetworkUSBSDMuxDevice/USBSDMuxDevice
    helios/alpha-group/RemoteNetworkInterface/USBNetworkInterface
  created: 2019-12-03 12:37:44.312045
  changed: 2023-09-14 13:08:37.743729
Acquired resource 'NetworkPowerPort' (helios/alpha-group/NetworkPowerPort/NetworkPowerPort):
  {'acquired': 'jenis',
   'avail': True,
   'cls': 'NetworkPowerPort',
   'params': {'extra': {'proxy': 'helios.swtech.dk', 'proxy_required': True},
              'host': 'http://192.168.0.2',
              'index': 0,
              'model': 'shelly_gen1'}}
Acquired resource 'NetworkService' (helios/alpha-group/NetworkService/NetworkService):
  {'acquired': 'jenis',
   'avail': True,
   'cls': 'NetworkService',
   'params': {'address': '192.168.0.75',
              'extra': {'proxy': 'helios.swtech.dk', 'proxy_required': True},
              'username': 'root'}}
Acquired resource 'USBMassStorage' (helios/alpha-group/NetworkUSBMassStorage/USBMassStorage):
  {'acquired': 'jenis',
   'avail': True,
   'cls': 'NetworkUSBMassStorage',
   'params': {'busnum': 1,
              'devnum': 7,
              'extra': {'proxy': 'helios.swtech.dk', 'proxy_required': True},
              'host': 'helios.swtech.dk',
              'model_id': 16449,
              'path': '/dev/sda',
              'vendor_id': 1060}}
Acquired resource 'USBNetworkInterface' (helios/alpha-group/RemoteNetworkInterface/USBNetworkInterface):
  {'acquired': 'jenis',
   'avail': True,
   'cls': 'RemoteNetworkInterface',
   'params': {'extra': {'proxy': 'helios.swtech.dk', 'proxy_required': True},
              'host': 'helios.swtech.dk',
              'ifname': 'enx00e04c680429'}}
Acquired resource 'USBSDMuxDevice' (helios/alpha-group/NetworkUSBSDMuxDevice/USBSDMuxDevice):
  {'acquired': 'jenis',
   'avail': True,
   'cls': 'NetworkUSBSDMuxDevice',
   'params': {'busnum': 1,
              'control_path': '/dev/sg0',
              'devnum': 7,
              'extra': {'proxy': 'helios.swtech.dk', 'proxy_required': True},
              'host': 'helios.swtech.dk',
              'model_id': 16449,
              'path': '/dev/sda',
              'vendor_id': 1060}}
Acquired resource 'USBSerialPort' (helios/alpha-group/NetworkSerialPort/USBSerialPort):
  {'acquired': 'jenis',
   'avail': True,
   'cls': 'NetworkSerialPort',
   'params': {'extra': {'path': '/dev/ttyUSB0',
                        'proxy': 'helios.swtech.dk',
                        'proxy_required': True},
              'host': 'helios.swtech.dk',
              'port': 39489,
              'speed': 115200}}
```

Emantor, 2023-09-14:
You'll want to remove the `USBMassStorage` match, the  `USBSDMux` can also be bound to by the `USBStorageDriver`. That's also what the (not very clear) error message is saying.

JohnTordur, 2023-09-14:
> You'll want to remove the `USBMassStorage` match, the `USBSDMux` can also be bound to by the `USBStorageDriver`. That's also what the (not very clear) error message is saying.

Thanks for getting back

I'm not quite sure how to achieve this. 
I see that USBStorageDriver can bind to  "USBSDMuxDevice", "NetworkUSBSDMuxDevice". 
But I'm not quite sure where and how to set that up. 
My best guess would be in env.yaml file, but not sure what fields to populate..

Emantor, 2023-09-14:
The current match block for your place looks like this:
```
*/alpha-group/*
```
Which means that you are including every Resource that matches this regex.
To remove this, delete the match:
```
labgrid-client -p jenis del-match */alpha-group/*
```
and add the matches you really require:
```
labgrid-client -p jenis add-match helios/alpha-group/NetworkPowerPort
labgrid-client -p jenis add-match helios/alpha-group/NetworkSerialPort
labgrid-client -p jenis add-match helios/alpha-group/NetworkService
labgrid-client -p jenis add-match helios/alpha-group/NetworkUSBSDMuxDevice
labgrid-client -p jenis add-match helios/alpha-group/RemoteNetworkInterface
```
Now the `USBSDMuxDevice` and the `USBMassStorage` no longer collide, we also don't match devices from a different exporter with the same group any longer.

Alternatively you can remove the `USBMassStorage` from the exporter yaml.

JohnTordur, 2023-09-14:
> The current match block for your place looks like this:
>
> ```
> */alpha-group/*
> ```
>
> Which means that you are including every Resource that matches this regex. To remove this, delete the match:
>
> ``` [...]

Thank you for the help, this pushed me in the right direction.

So I assumed that I had to set up `USBMassStorage` resource for the SD Mux in addition to the `USBSDMuxDevice` resource. 

Removing the `USBMassStorage` solved it.

closed 2023-09-14 by JohnTordur