labgrid-project/labgrid #1289: Add remote CAN resource

stoupa-cz, 2023-10-23:
With local can resources in place (see #1287 ), we can add remote access with NetworkCANPort. We discussed the approach at EOSS2023. Pengutronix guys suggest bridging by one of these tools:

- [cannelloni](https://github.com/mguentner/cannelloni)
- [socketcand](https://github.com/linux-can/socketcand)

The idea of bridging is similar to NetworkSerialPort. We will start a daemon on an exporter. A client will create a local `vcan<x>` interface by connecting to the daemon.

I don't know about these projects, so if there is any recommendation which is better to start with, please let me know.

jluebbe, 2023-10-24:
@marckleinebudde Can you give some guidance here?

stoupa-cz, 2025-07-23:
We are again on that topic. I briefly checked both projects above, and both still seem fine. I'd be happy to hear your input on the selection and suitability for labgrid (@marckleinebudde ?).

CC @ca-ja