labgrid-project/labgrid #1313: ShellDriver: Idle state discovery using newline cannot recover shell

fanuware, 2024-01-03:
The ShellDriver is using a newline `self.console.sendline("")` to discover the idle state. A newline cannot resolve issues, particularly in situations where a shell prompt is left with an open quote or when another command has not yet been terminated.

Instead of newline we might want to consider using Ctrl-C `self.console.sendline("\x03")` which acts similar as idle state discovery but additionally also aborts (SIGINT) the current shell process.

Bastian-Krause, 2024-01-04:
If the shell is busy waiting for a command to terminate, it's obviously not idle. The "open quote" issue should not happen in the first place. Sending Ctrl-C seems pretty intrusive.

What's the concrete issue your facing that makes you want to send Ctrl-C?

fanuware, 2024-01-05:
There are cases where a previous test run fails due to a timeout and the Shell is left in a "open quote" state.
When starting a new test run how to recover such a non-idle state?

Bastian-Krause, 2024-01-05:
Note: there's also `console.sendcontrol("c")`.

fanuware, 2024-01-08:
Thank you for the helpful input. Indeed, using the underlying ConsoleProtocol allows aborting a process as expected.

However, should it not be the concern of the ShellDriver to fully manage the ConsoleProtocol and also be able to recover from such a case?

Bastian-Krause, 2024-03-14:
labgrid assumes exclusive access to the shell, so no intrusive recovery methods are implemented.

closed 2024-03-14 by Bastian-Krause