labgrid-project/labgrid #1325: SSHDriver: Ubuntu 20.04 Clients getting stuck when trying to ssh into target via exporter proxy

ep1cman, 2024-02-02:
We currently have the following setup:

- A remote exporter running in isolated mode, with all device attached to it
- Labgrid clients interacting with the devices by SSH proxing via the remote exporter machine

On Ubuntu 22.04 (OpenSSH 8.9) and MacOS (OpenSSH ﻿9.3) clients. This works as expected.

On Ubuntu 20.04 (OpenSSH 8.2) clients however stall when attempting to run:
```
labgrid-client -p PLACE_NAME ssh
```

Investigations reveal it is getting stuck here: https://github.com/labgrid-project/labgrid/blob/2d6f93c0e298442453603f2fd5430da37606da09/labgrid/driver/sshdriver.py#L151

Commenting out the following line seems to resolve the issue

https://github.com/labgrid-project/labgrid/blob/2d6f93c0e298442453603f2fd5430da37606da09/labgrid/driver/sshdriver.py#L142

Logs from a working machine:
```

Bastian-Krause, 2024-02-06:
Might be related to #1288.

ep1cman, 2024-02-07:
Ah yes, you are correct. Looks like we had deviated from the upstream code and this didn't get pulled in. Seems to work ok now, if I face further issues I will re-open

closed 2024-02-07 by ep1cman