labgrid-project/labgrid #1333: test_shell script returns [Errno13] Permission denied when trying to access serial port

gcsam11, 2024-02-23:
I'm connected through SSH to a Raspberry Pi 4 where the serial port are registered as /dev/ttyX (using 0). The user is already in groups tty and dialout and it's also important to note that while trying sudo lsof /dev/tty0 i get error `lsof: WARNING: can't stat() fuse.gvfsd-fuse file system /run/user/1000/gvfs 
Output information may be incomplete.`
This is my local.yaml file: 
```yaml
targets:
  main:
    resources:
      RawSerialPort:
        port: "/dev/tty0"
    drivers:
      ManualPowerDriver:
        name: "example"
      SerialDriver: {}
      ShellDriver:
        prompt: 'gonca@\w+:[^ ]+ '
        login_prompt: 'login: '
        username: 'gonca'
        password: '-----'
```

Emantor, 2024-02-23:
Hm, at least on my CM4 powered device, `udevadm info` identifies `tty0` as a virtual port. The platform serial port is exported as `/dev/ttyAMA0`. Can you check this on your device? And please post the output of:
`ls /dev/tty*`
`udevadm info /dev/tty0`
Thanks.

gcsam11, 2024-02-28:
You are correct. `tty0` is indeed a virtual port and I noticed after I had a ttyS0 which is the correct one. After swapping this in the yaml file, I now get error `ERROR test_shell.py::test_shell - serial.serialutil.SerialException: Could not open serial port /dev/ttyS0: Could not configure port: (5, 'Input/output error')`.

gcsam11, 2024-02-28:
[image]
[image]
[image]
[image]

Emantor, 2024-02-28:
Your device seems to be available as `/dev/ttyAMA2`, this also has the correct permissions for your user to open it.

Bastian-Krause, 2024-03-14:
@gcsam11 Did `/dev/ttyAMA2` solve the issue for you?

closed 2024-03-14 by Bastian-Krause