labgrid-project/labgrid #1341: SSHDriver - '_scp_supports_explicit_sftp_mode' fails

krevsbech, 2024-03-21:
On consecutive scp transfers to target the:
`_scp_supports_explicit_sftp_mode` fails on **second** "put".

```
def _scp_supports_explicit_sftp_mode(self):
>       major, minor = self._ssh_version
E       ValueError: not enough values to unpack (expected 2, got 0)
```
It has to do with the`@cached_property` introduced in commit: `2c062e3ac6bb55308ddc9b2bdd3602b09f50f3fe`
Hence reverting this commit fixes the problem.
I am not sure it is a general thing, hence my settings for the driver:
```
- SSHDriver:
    name: 'ssh_root'
    bindings:
      networkservice: 'ssh_root_service'
    keyfile: '~/.ssh/target_specific_key'
    explicit_sftp_mode: true
```

jluebbe, 2024-03-21:
Ah, it caches the map object (which is an iterator). Let me try to make a fix.

jluebbe, 2024-03-21:
@krevsbech Does https://github.com/labgrid-project/labgrid/pull/1342 help?

krevsbech, 2024-03-21:
@jluebbe Yes that fixes the problem. Thanks for the fast handling. I was in doubt what was the underlying problem, and thought it was a special case:)