labgrid-project/labgrid #1350: bootstrap without USB

sjg20, 2024-04-09:
I would like to bootstrap any device, not just one with a USB bootrom.

For example, for rpi3 (Raspberry Pi) I want to:

- switch the sdwire to 'host'
- mount the first partition
- update the 'config.txt' and kernel.img files
- unmount it
- switch the sdwire to 'dut'

After that, cycling power/ reset should boot the new bootloader.

It looks like this might be possible with a new driver that implements the BootstrapProtocol. There is already USBStorageDriver for accessing the filesystem, but it seems that much of the above could be handled automatically, with perhaps an RpiDriver (with BootstrapProtocol) that does the sd-mux and USBStorageDriver.write_files()

It might be necessary to add some yaml info to RpiDriver to tell it which rpi version it is (e.g. 3), but perhaps it can figure out the sdwire and storage itself?

Emantor, 2024-04-09:
We usually advice to handle more complicated sequences which are not restricted to one resource via a strategy, these can also be loaded by the labgrid-client via an environment file. See [Using a strategy](https://labgrid.readthedocs.io/en/latest/getting_started.html#using-a-strategy).

sjg20, 2024-04-13:
OK I am trying that and will come back on this

sjg20, 2024-04-26:
This seems to work OK. I need a few other features but will raise issues / discussions for those

closed 2024-04-26 by sjg20