labgrid-project/labgrid #1352: docs: WARNING: autodoc: failed to import module 'eaton' from module ...

sjg20, 2024-04-10:
I am seeing warnings when building the docs:

``
cd doc
make html
...
WARNING: autodoc: failed to import module 'eaton' from module 'labgrid.driver.power'; the following exception was raised:
No module named 'pysnmp'
WARNING: autodoc: failed to import module 'poe_mib' from module 'labgrid.driver.power'; the following exception was raised:
No module named 'pysnmp'
WARNING: autodoc: failed to import module 'raritan' from module 'labgrid.driver.power'; the following exception was raised:
No module named 'pysnmp'
WARNING: autodoc: failed to import module 'siglent' from module 'labgrid.driver.power'; the following exception was raised:
No module named 'vxi11'
WARNING: autodoc: failed to import module 'tplink' from module 'labgrid.driver.power'; the following exception was raised:
No module named 'kasa'
WARNING: autodoc: failed to import module 'snmp' from module 'labgrid.util'; the following exception was raised:
No module named 'pysnmp'
```

Emantor, 2024-04-11:
Looks like we have a whole bunch of modules which were never added to the pyproject.toml file but are required for automatic doc generation. We'll need to check whether they are required or if the warnings can be safely ignored.

Bastian-Krause, 2024-04-15:
> Looks like we have a whole bunch of modules which were never added to the pyproject.toml file but are required for automatic doc generation. We'll need to check whether they are required or if the warnings can be safely ignored.

Huh? All these modules are referenced in the pyproject.toml in the "dev" extra.

@sjg20 If you want to have autodoc functionality, follow the [steps in the docs](https://labgrid.readthedocs.io/en/latest/development.html#building-the-documentation ), but use `pip install -e ".[dev]"` instead of `pip install -e ".[doc]"`. A PR adding this to this docs section would be appreciated :)

sjg20, 2024-04-20:
OK, will do. I also hit this problem:

https://github.com/readthedocs/readthedocs.org/issues/9038

Python seems to be a disaster at the moment

commit f2c25686 (2024-04-22): doc: Install [dev] so that docs can be built

The 'make html' command fails with the [doc] build. Update the
documentation to show how to resolve this.

Fixes: #1352
PR #1367 (2024-04-22): doc: Install [dev] so that docs can be built
The 'make html' command fails with the [doc] build. Update the documentation to show how to resolve this.

Fixes #1352
closed 2024-04-22 by jluebbe