labgrid-project/labgrid #1353: docs: Failed to build PDF

sjg20, 2024-04-10:
`make help` shows:
```
  latexpdf    to make LaTeX and PDF files (default pdflatex)
```

but using `make latexpdf` fails. In fact it seems to hang in latex somewhere:

```
[25] [26] [27] [28]
Chapter 4.
[29] [30] [31] [32] [33] [34] [35] [36] [37] [38] [39] [40]
Chapter 5.

! LaTeX Error: Unknown graphics extension: .svg.

See the LaTeX manual or LaTeX Companion for explanation.
Type  H <return>  for immediate help.
 ...                                              
                                                  
l.2628 ...dth=0.500\linewidth]{{config_graph}.svg}
                                                  
? 

! LaTeX Error: Unknown graphics extension: .svg.

See the LaTeX manual or LaTeX Companion for explanation.
Type  H <return>  for immediate help.
 ...                                              
                                                  
l.2628 ...dth=0.500\linewidth]{{config_graph}.svg}
                                                  
?
```

jluebbe, 2024-04-11:
We could also just disable PDF completely. Is there a need for them with online and offline HTML docs?

Bastian-Krause, 2024-04-12:
Btw, RTD also generates pdf documentation for releases: https://readthedocs.org/projects/labgrid/downloads/

sjg20, 2024-04-13:
PDFs are helpful for printing a proper doc with page numbers, TOC, index

sjg20, 2024-05-02:
BTW do web browsers support SVG now? I certainly see some pages that contain them