labgrid-project/labgrid #1354: Confused by crossbar / router / coordinator

sjg20, 2024-04-10:
Reading the Overview I am confused between the terms crossbar / router / coordinator.  What is the difference between these? Could the overview introduce each term one by one and explain it?

Would you be open to some wording / highlighting changes in the docs? There are some minor inconsistencies which can make things confusing.

Emantor, 2024-04-11:
- crossbar: WAMP (Web Application Message Protocol) router for the labgrid distributed infrastructure.
- coordinator: Process Spawned by the crossbar to handle requests from client and exporters. Stores place information and regularly pings exporters to ensure they are still responsive. Also provides aggregation of exporter information.

Early on we decided on crossbar for the distributed infrastructure, we want to replace it with gRPC eventually. Crossbar is no longer well maintained.

sjg20, 2024-04-20:
OK thank you

closed 2024-04-20 by sjg20