labgrid-project/labgrid #1368: How to power off a board after quitting the console?

sjg20, 2024-04-21:
I have something like this:

```
LG_CROSSBAR=ws://kea:20408/ws labgrid-client  -c /vid/software/devel/ubtest/lab/env_rpi_try.cfg  -p rpi3 -s uboot console
```

This powered on the board and starts U-Boot. When I quite labgrid (with Ctrl-\ twice) I want it to power the board off again. I suppose I am saying that I want it to select the UBootStrategy 'off' state before exiting, e.g. adding a new arg like '-e off'

How would I implement that? I tried adding this logic to the end of session.loop.run_until_complete(coro) in client.py but it seems quit that without returning

Emantor, 2024-05-02:
I'd not handle this in pure python code and rather employ a shell wrapper for this. Note also that `Crtl-\` twice is not the correct way to quit microcom and produces a core dump.

sjg20, 2024-05-02:
@Emantor Thanks for the info. I am quite reluctant to add shell scripts around the labgrid-client invocation. It is labgrid that knows how to power a particular board off. Building all the infrastructure for knowing how to do this on each board sounds like writing bits of labgrid all over again in shell script, at least to me.

What is the correct way to exit minicom? I don't see it in the docs, but might be missing it.