labgrid-project/labgrid #1374: Controlling a strategy at runtime

sjg20, 2024-04-26:
I would like to be able to tell the U-Boot strategy what to do at runtime, e.g.:

1. just start up the board and connect a console
2. write U-Boot to the board, then do 1
3. build U-Boot and then do 2

At the moment I have implemented this with a new --variable (-V) argument to labgrid-client, with these variables available for use by strategies.

By far the most common use case I have is to build, write and start U-Boot on a board (case 3). This is:
```
labgrid-client -c env_rpi_try.cfg -V do-bootstrap 1 -V do-build 1 -p opi_pc -s start console 
```
To just write the already-built U-Boot (case 2):
```
labgrid-client -c env_rpi_try.cfg -V do-bootstrap 1 -V do-build 0 -p opi_pc -s start console 
```
To just start with whatever the board has (case 1):
```
labgrid-client -c env_rpi_try.cfg -V do-bootstrap 0 -V do-build 0 -p opi_pc -s start console 
```

Does this seems like a reasonable feature? Is there an existing mechanism to control strategies from the cmdline?

Emantor, 2024-04-26:
States of a strategy can already be requested with the `-s` command line argument, AFAIR the strategy needs to implement the `transition()` function.

sjg20, 2024-04-26:
This is not actually about the state of a strategy. It is about the states that it should go through and what exactly they should do. I have been experimenting with these for UBootStrategy:

        unknown: State is not known
        off: Power is off
        bootstrap: U-Boot has been written to the board
        start: Board has started booting
        uboot: Board has stopped at the U-Boot prompt
        shell: Board has stopped at the Linux prompt

Note that 'start' is useful since it avoid suppressing the start-up message, and shows the output even if the board fails to boot to a prompt.

The 'bootstrap' state is selected if do-bootstrap is true., otherwise it is bypassed.

The bootstrap state can be told to build U-Boot, or just to use the existing build

sjg20, 2024-04-26:
Are you open to enhancements to UBootStrategy?

The -s command allows one state to be selected, but in this case I want to bring up the console and control which states are used to get there. This avoids having to run labgrid-client multiple times to get to the console.