labgrid-project/labgrid #1381: Absolute paths

sjg20, 2024-04-29:
Why does labgrid use relative paths for imports? When using code inside and outside labgrid, it requires transformation.

Would it not be easier to use absolute imports everywhere?

Emantor, 2024-05-02:
Relative paths are used since the imports key is usually used to import files for the local test definition which should live in the same repository. Our usecase for this is usually a customer and board specific strategy for the target hardware. For global additions to labgrid I would recommend not using the imports key and rather use your own python module installed into the same virtualenv.

sjg20, 2024-05-02:
@Emantor What is the 'import key'? I had thought that Python encourages absolute imports. I have tended to uses absolute imports in my own Python programs and have not hit any problems with that

Emantor, 2024-05-03:
The device configuration YAML contains an [imports](https://labgrid.readthedocs.io/en/latest/man/device-config.html#imports) key which can be a list of python files that are imported before target creation.