labgrid-project/labgrid #1383: ser2net fails on place acquire

colin-foster-in-advantage, 2024-04-29:
I have a setup where I'm running Ubuntu 24.04 as a labgrid-exporter for a USB serial port. Whenever I try to acquire the resources, I get a failure when trying to start the ser2net subprocess

```
2024-04-29T15:14:26 marked as broken: start failed                                                                                                                                   
1969-12-31T18:00:00                                                                                                                                                                  
{'cls': 'NetworkSerialPort', 'params': OrderedDict({'host': 'lg-omap', 'port': None, 'speed': 115200, 'extra': {'path': '/dev/ttyUSB0', 'proxy_required': False, 'proxy': 'lg-omap'}}
), 'acquired': '<broken>', 'avail': True}                                                                                                                                            
{'cls': 'NetworkSerialPort', 'params': OrderedDict({'host': 'lg-omap', 'port': 40999, 'speed': 115200, 'extra': {'path': '/dev/ttyUSB0', 'proxy_required': False, 'proxy': 'lg-omap', 'broken': 'start failed'}}), 'acquired': '<broken>', 'avail': False}                                                                                                                
2024-04-29T15:14:26 ExporterSession.onUserError(): "CalledProcessError: Command '['/usr/sbin/ser2net', '-v']' returned non-zero exit status 1."                                      
Traceback (most recent call last):                                                                                                                                                   
  File "/opt/labgrid-exporter/venv_labgrid/lib/python3.12/site-packages/txaio/aio.py", line 487, in done                                                                             
    res = f.result()                                                                                                                                                                 
          ^^^^^^^^^^                                                                                                                                                                 
  File "/opt/labgrid-exporter/venv_labgrid/lib/python3.12/site-packages/labgrid/remote/exporter.py", line 797, in acquire                                                            
    resource.acquire(place_name)                                                                                                                                                     
  File "/opt/labgrid-exporter/venv_labgrid/lib/python3.12/site-packages/labgrid/remote/exporter.py", line 170, in acquire                                                            
    self.poll()                                                                                                                                                                      
  File "/opt/labgrid-exporter/venv_labgrid/lib/python3.12/site-packages/labgrid/remote/exporter.py", line 139, in poll                                                               
    self.start()                                                                                                                                                                     
  File "/opt/labgrid-exporter/venv_labgrid/lib/python3.12/site-packages/labgrid/remote/exporter.py", line 113, in start                                                              
    self._start(start_params)                                                                                                                                                        
  File "/opt/labgrid-exporter/venv_labgrid/lib/python3.12/site-packages/labgrid/remote/exporter.py", line 231, in _start                                                             
    _, _, version = str(subprocess.check_output([self.ser2net_bin,'-v'])).split(' ')                                                                                                 
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^                                                                                                             
  File "/usr/lib/python3.12/subprocess.py", line 466, in check_output                                                                                                                
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,                                                                                                                 
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^                                                                                                                 
  File "/usr/lib/python3.12/subprocess.py", line 571, in run                                                                                                                         
    raise CalledProcessError(retcode, process.args,                                                                                                                                  
subprocess.CalledProcessError: Command '['/usr/sbin/ser2net', '-v']' returned non-zero exit status 1.                                                                                
```

ser2net seems to exist at `/usr/sbin/`, and the version should properly return:

```
(venv_labgrid) $ which ser2net
/usr/sbin/ser2net                                                
(venv_labgrid) $ ser2net -v   
ser2net version 4.6.0                                            
```

Maybe there is a version dependency with txaio? Or there's something simple I'm missing. My current setup is just running these commands as a normal user for now - no systemd involvement.

colin-foster-in-advantage, 2024-04-30:
Figured it out. ser2net version 4.6.0 seems to report an error when invoked with -v. That causes the exporter to fail.

I'll raise an issue with them as well.

Emantor, 2024-04-30:
We could work around this by using a different check and ignoring the error code. Not sure how many people will hit this error, we should document it either way.

PR #1395 (2024-05-07): exporter: check ser2net version in exporter
The following ser2net versions return nonzero even on a successful use of the -v flag to get the version:

v4.4.0
v4.5.0
v4.5.1
v4.6.0
v4.6.1

This was fixed in v4.6.2, which was only recently released. Modify _start() so that it uses subprocess.run() to capture stdout and the returncode, then check against the version and effectively ignore the return value if there's a match. If the ser2net version is one of the unaffected ones (e.g. v4.6.2) and still returns nonzero, then raise an ExporterError.

Tested locally by installing in a venv and running  [... cut]
closed 2024-05-07 by jluebbe