labgrid-project/labgrid #1384: Add ability to jump in networkresource

flxzt, 2024-04-30:
It would be very nice to have the ability to specify SSH jumps for `NetworkResource`s.

The main use-case would be where DUT's only have exposed networking to another specific device and not reachable directly from the client or exporter.

Would adding a `jumps` field to `NetworkResouce` that contains a list of dicts specifying the jumps make sense?

Emantor, 2024-04-30:
Labgrid already supports jumping over the exporter to the DUT for NetworkServiceExports, this is done by using the `proxymanager` class inside of labgrid. This is done when the `address` includes a IPv6 link-local address with interface extension, i.e. `fe80::1%dut` where the interface is named `dut` and the IPv6 link local address is `fe80::1`. You can also force a different proxy by importing `proxymanager` and using the `force_proxy()` function.

The other choice is to ship a correctly configured SSH configuration which includes the correct `ProxyJump` depending on the IP address.

flxzt, 2024-04-30:
No, I meant specifying a network connection where you essentially have two targets, one is the DUT you want to connect to, but it might have a specific network connection and configuration where it is only available through another device which is part of the test setup. (like for example: devices with network connections only available through a PCI backplane)

I think this is currently possible by managing such a jump as a ssh configuration, but this would need to live on the client which I want to avoid.

For example it currently can work like this:

exporter config

```
NetworkService
    address: dut-through-controller
    username: 'root'
```

and the client-side ssh config

```
Host testcontroller
    HostName 10.0.1.23
    User root

Host dut-through-controller
    HostName 192.168.0.2
    User root
    ProxyJump controller
```

I'd like it to become something like this:

exporter config

```
NetworkService
    address: 192.168.0.2
    username: 'root'
    jumps:
    - address: 10.0.1.23
      username: 'root'
```

Emantor, 2024-05-02:
> No, I meant specifying a network connection where you essentially have two targets, one is the DUT you want to connect to, but it might have a specific network connection and configuration where it [...]

The problem here is that proxymanager only allows setting a proxy globally (since it's requirements arose from the client CLI proxy requirement).

> As you wrote this is currently possible by managing such a jump as a ssh configuration, but this would need to live on the client which I want to avoid because then each developer would need to specify this themselves.

You can ship the SSH configuration with your test files and provide a wrapper script to access the second device. However here the ssh `-F` option does not merge with the global configuration.

I am hesitant to add this because this duplicates functionality already found within SSH. IMO rolling out a working SSH config to developers would be a preferable option to handling this in labgrid.

flxzt, 2024-05-16:
Hm, I can see your point about avoiding replicating features already found in SSH but I would also argue that this separation of configuration from the client is one of the main features of labgrid.

> The problem here is that proxymanager only allows setting a proxy globally (since it's requirements arose from the client CLI proxy requirement).

I'd put this feature into the connection manager class because there could be both - a proxy over the exporter which is managed in proxymanager *and* additional jumps to the target which are different per connection.

I believe it could be implemented by adding `jumps` to all resources which derive `NetworkResource`. 
Then, in `command_prefix` the `SSHConnection` can be supplied with these additional arguments.