labgrid-project/labgrid #1386: labgrid-client console: add escape sequence to call other client commands

khilman, 2024-05-02:
Story: I'm using `labgrid-client -p <place> console` to connect to a place.  From **within the console window**, I want to trigger a power on/off/cycle (e.g. the equivalent of callling `labgrid-client -p <place> power cycle` from another shell.)  

This request is inspired by similar feature in `conmux` where for any given console, you can use an escape sequence (e.g. `~$` in conmux) which can commands on the host where the conmux server (same as labgrid-exporter) is running.

Emantor, 2024-05-03:
I had the same idea as well and do have a prototype somewhere which implements a labgrid-client CLI which takes commands and executes them without closing the client connection. To implement this we would either need to write our own RFC2711 client with an escape sequence for this CLI or wire up the RFC2711 client that is AFAIR included within pyserial.

jluebbe, 2024-05-03:
I'd find that useful as well, but it would probably mean using an internal telnet/RFC2711 client instead of microcom. Also, it would lend itself to supporting hot keys as well (e.g. CTRL-<something> to trigger a power cycle or a strategy state). So far we've shied away from that due to the possible maintenance required.

khilman, 2024-05-03:
You can see some discussion of it in the conmux docs here (search for `escape`):
https://autotest.readthedocs.io/en/latest/main/remote/Conmux-OriginalDocumentation.html

And the code is here: https://github.com/autotest/autotest/tree/master/conmux
Though TBH, I haven't ever looked at this code (it's in perl) so I won't pretend to understand how it works.  

More specifically, it seems like it's here where the conmux escape sequence `~$` is handled:
https://github.com/autotest/autotest/blob/0862aa74389bf434533697a9d970cf91e03b09ce/conmux/conmux#L1049

ukleinek, 2025-04-09:
Just a crazy idea: Extend RFC2217 with a command (set?) to do labgrid stuff? (Or rather extend telnet as several other RFCs (e.g. RFC2217, but not RFC2711 😃) did.