labgrid-project/labgrid #1390: Recovery and reset on the same relay

sjg20, 2024-05-03:
For tegra boards (and also Samsung) I need to a reset sequence like this:

- assert reset
- assert recovery
- de-assert reset
- de-assert recovery

For some boards I have reset and recovery on the same HIDrelay. How do I tell labgrid which is which?

My exports are:
```
  HIDRelay:
    ## reset
    ## usbrelay 0 2
    ##serial: 6QMBS
    index: 2
    usbrelay: True
    match:
      '@ID_PATH': 'pci-0000:00:14.0-usb-0:10.1.2'

  HIDRelay:
    ## recovery
    ## usbrelay 0 3
    ##serial: 6QMBS
    index: 3
    usbrelay: True
    match:
      '@ID_PATH': 'pci-0000:00:14.0-usb-0:10.1.2'
```

and I'm not sure what to put in the env:

```
  tk1:
    resources:
      RemotePlace:
        name: tk1

    drivers:
      AlwaysPowerDriver: {}
      SerialDriver: {}
      DigitalOutputResetDriver: {}
      DigitalOutputRecoveryDriver: {}       -- added by me, a 'RecoveryProtocol' driver
      HIDRelayDriver: {}
```

It seems that labgrid assumes that DigitalOutputResetDriver refers to the only digital output available. But what if there are two? I don't see how to make this connection in the various configuration files.

sjg20, 2024-05-10:
Any thoughts on this one?

The place only shows one rely:
```
Place 'tk1':
  matches:
    kea/tk1/*
  acquired: ellesmere/sglass
  acquired resources:
    kea/tk1/NetworkHIDRelay/HIDRelay
    kea/tk1/NetworkSerialPort/NetworkSerialPort
    kea/tk1/NetworkTegraUSBLoader/TegraUSBLoader
  created: 2024-05-03 12:08:38.000100
  changed: 2024-05-10 10:31:59.389736
```

When I try to use this place, I get:

```
LG_CROSSBAR=ws://kea:20408/ws labgrid-client -vv -c /vid/software/devel/ubtest/lab/env_rpi_try.cfg -V do-bootstrap 1 -V do-build 1 -V do-send 1 -p tk1 -s start -a console 
INFO               root: Selected role tk1 from configuration file
Traceback (most recent call last):
  File "/vid/software/devel/ubtest/labgrid/labgrid/target.py", line 373, in bind_driver
    self.get_driver(requirement, name=supplier_name, activate=False),
  File "/vid/software/devel/ubtest/labgrid/labgrid/target.py", line 240, in get_driver
    return self._get_driver(cls, name=name, resource=resource,
  File "/vid/software/devel/ubtest/labgrid/labgrid/target.py", line 189, in _get_driver
    raise NoDriverFoundError(
labgrid.exceptions.NoDriverFoundError: no DigitalOutputProtocol driver found in Target(name='tk1', env=Environment(config_file='/vid/software/devel/ubtest/lab/env_rpi_try.cfg'), var_dict=None)

The above exception was the direct cause of the following exception:
...
```

So it seems to see only one of the relay channels??

sjg20, 2024-06-15:
I have another board which needs this (Beagleplay).

Undef 'Exporter Configuration' I can see how to create a group., but I am not sure how to apply this here:

```
  HIDRelay:
    ## reset
    ## usbrelay 2 6
    ## hub a 10
    ## serial: 7QMBS
    index: 6
    match:
      'ID_PATH': 'pci-0000:00:14.0-usb-0:10.4.2'

  HIDRelay:
    ## recovery (USR1)
    ## usbrelay 2 7
    ## hub a 10
    ## serial: 7QMBS
    index: 7
    match:
      'ID_PATH': 'pci-0000:00:14.0-usb-0:10.4.2'
```

They both have the same match string, of course...

BTW I have some ideas on how to improve the configuration, if anyone is interested.

Emantor, 2024-06-17:
Labgrid supports this, but you have to use the `labgrid-client add-named-match` for resources and than manually specify the bindings for both IODrivers. Part of it is explained in the [labgrid-client man page](https://labgrid.readthedocs.io/en/latest/man/client.html#adding-named-resources) and another part in [this](https://github.com/labgrid-project/labgrid/discussions/1259) discussion.

sjg20, 2025-02-15:
I managed to get this working, with the below:

```
play:
  location: 'lab'
  USBSerialPort:
    ## hub
    match:
      '@ID_PATH': pci-0000:03:00.2-usb-0:2.4.3:1.0

  USBSDWireDevice:
    ## hub
    match:
      ID_SERIAL_SHORT: da69

  reset:
    ## reset
    ## usbrelay 3 5
    ## hub e 11
    ##serial: 8QMBS (hub e11)
    cls: HIDRelay
    index: 5
    match:
      ID_PATH: pci-0000:03:00.2-usb-0:3.4.3

  recovery:
    ##recovery (USR1)
    ##usbrelay 2 6
    ##hub a 10
    ##serial: 7QMBS
    cls: HIDRelay
    index: 6
    match:
      ID_PATH: pci-0000:00:14.0-usb-0:10.4.2
```

The environment fragment is:
```
  play:
    resources:
      RemotePlace:
        name: play

    features:
      - flash
      - build

    drivers:
    - SerialDriver:
        txdelay: 0.01
    - USBSDWireDriver: {}
    - AlwaysPowerDriver: {}
    - HIDRelayDriver:
        name: reset_driver
        bindings:
          relay: reset
    - HIDRelayDriver:
        name: recovery_driver
        bindings:
          relay: recovery
    - DigitalOutputResetDriver:
        name: reset_output
        bindings:
          output: reset_driver
    - DigitalOutputRecoveryDriver:
        name: recovery_output
        bindings:
          output: recovery_driver
    - USBStorageDriver: {}
    - UBootDriver:
        prompt: '=>'
        autoboot: "Press SPACE to abort autoboot"
        login_timeout: 15
        interrupt: ' '
    - ShellDriver:
        prompt: 'root@\w+:[^ ]+ '
        login_prompt: ' login: '
        username: 'root'
    - UBootStrategy:
        recovery_reset: True
    - UBootProviderDriver:
        board: am62x_beagleplay_a53
        board_extra: am62x_beagleplay_r5
        bl31: /scratch/sglass/trusted-firmware-a/build/k3/lite/release/bl31.bin
        tee: /scratch/sglass/optee_os/out/arm-plat-k3/core/tee-raw.bin
        binman_indir: /scratch/sglass/ti-linux-firmware
    - UBootWriterDriver:
        method: ti,am625
```

  play:
    resources:
      RemotePlace:
        name: play

    features:
      - flash
      - build

    drivers:
    - SerialDriver:
        txdelay: 0.01
    - USBSDWireDriver: {}
    - AlwaysPowerDriver: {}
    - HIDRelayDriver:
        name: reset_driver
        bindings:
          relay: reset
    - HIDRelayDriver:
        name: recovery_driver
        bindings:
          relay: recovery
    - DigitalOutputResetDriver:
        name: reset_output
        bindings:
          output: reset_driver
    - DigitalOutputRecoveryDriver:
        name: recovery_output
        bindings:
          output: recovery_driver
    - USBStorageDriver: {}
    - UBootDriver:
        prompt: '=>'
        autoboot: "Press SPACE to abort autoboot"
        login_timeout: 15
        interrupt: ' '
    - ShellDriver:
        prompt: 'root@\w+:[^ ]+ '
        login_prompt: ' login: '
        username: 'root'
    - UBootStrategy:
        recovery_reset: True
    - UBootProviderDriver:
        board: am62x_beagleplay_a53
        board_extra: am62x_beagleplay_r5
        bl31: /scratch/sglass/trusted-firmware-a/build/k3/lite/release/bl31.bin
        tee: /scratch/sglass/optee_os/out/arm-plat-k3/core/tee-raw.bin
        binman_indir: /scratch/sglass/ti-linux-firmware
    - UBootWriterDriver:
        method: ti,am625
```

closed 2025-02-15 by sjg20