labgrid-project/labgrid #1392: write_file does not properly escape file names

fhars, 2024-05-06:
Somewhere along the way from `write_files` through `ProcessWrapper`, `subprocess.Popen`, `ssh` to the final `cp` command, there seems to be a shell (I suspect ssh) that expands file names, so things fail if filenames contain spaces. Or other interesting things:

```
$ date; labgrid-client -c test-config/remote.yaml  -p my_place write-files -t '/; touch I_Was_Here' 'fnord'; ls -l ~/I_Was_Here
Mo 6. Mai 13:51:07 CEST 2024
Selected role main from configuration file
-rw-r--r-- 1 hars hars 0  6. Mai 13:51 /home/hars/I_Was_Here
```

Emantor, 2024-05-06:
Yep, looks like we are missing a `shlex.quote()`, we should probably place this in the `USBStorageDriver` to ensure that passed in folders are properly escaped.

fhars, 2024-05-06:
From ssh(1):

> If supplied, the arguments will be appended to the command, separated by spaces, before it is sent to the server to be executed.

Emantor, 2024-05-06:
Yep, please send a PR if you find the time.