labgrid-project/labgrid #1415: Timeout Parameter for "labgrid-client reserve"

hbml, 2024-05-31:
In our GitLab CI workflows, we encounter issues with tests running longer than expected, leading to job timeouts and unintended interruptions. We propose adding a timeout parameter to the `labgrid-client reserve` command to ensure the test environment is released and returned to a known-good state even if a timeout occurs.

Proposed Solution:
Add a timeout parameter to the `labgrid-client reserve` command (e.g., 1 hour) to ensure ample time to run tests and restore the test environment.

Thank you for considering this enhancement.

jluebbe, 2024-06-28:
I'd like to understand your case better. Other running test keep the places locked longer than expected? If those other tests are stalled, perhaps it would be better to abort that test run and let the waiting reservation continue?

We're currently reworking the coordinator to use gRPC: https://github.com/labgrid-project/labgrid/discussions/1417
Based on that code, it should be easy to abort after a timeout expires: https://github.com/labgrid-project/labgrid/blob/grpc/labgrid/remote/client.py#L1443 (in the _wait_reservation function).

hbml, 2024-08-05:
Sorry for the late answer, thank you for your response and the detailed information.

To address our issue, we have added a timeout command at the beginning of the reserve command in our GitLab CI pipelines. This solution has effectively resolved our problem without requiring any upstream changes.

We appreciate your support and the suggestions you provided. 

BR

closed 2024-08-05 by hbml