labgrid-project/labgrid #1420: grpc error

sjg20, 2024-06-12:
This is what I got trying the new grpc branch:
```
 HDCTOOLS=/vid/software/devel/ubtest/standalone-hdctools LG_CROSSBAR=ws://kea:20408/ws labgrid-exporter ../cfg/export_kea.yaml
exporter name: kea
exporter hostname: kea
resource config file: ../cfg/export_kea.yaml
('rendered',
 '\n'
 '\n'
 '\n'
 '\n'
 '\n'
 '\n'
 '\n'
 'rpi3:\n'
 "  location: 'lab'\n"
 '  USBSerialPort:\n'
 '\n'
 '\n'
 '    match:\n'
 "      '@ID_PATH': 'pci-0000:03:00.2-usb-0:5.2.4:1.0'\n"
 '\n'
 '  USBSDWireDevice:\n'
 '\n'
 '\n'
 '    match:\n'
 '\n'
 "      'ID_SERIAL_SHORT': 'sdwire-18'\n"
 '\n'
 '  YKUSHPowerPort:\n'
 '\n'
 "    serial: 'YK17698'\n"
 '    index: 1\n'

...

add resource rock2/YKUSHPowerPort: YKUSHPowerPort/OrderedDict([('serial', 'YK17698'), ('index', 3)])
YKUSHPowerPortExport(data={'avail': False, 'cls': 'NetworkYKUSHPowerPort', 'params': OrderedDict(), 'acquired': None}, host='kea', proxy='kea', proxy_required=False, local=YKUSHPowerPort(target=None, name=None, state=<BindingState.idle: 0>, avail=True, serial='YK17698', index=3), local_params=OrderedDict([('serial', 'YK17698'), ('index', 3)]), start_params=None)
{'cls': 'NetworkYKUSHPowerPort', 'params': OrderedDict(), 'acquired': None, 'avail': False}
encoded resource {
  path {
    group_name: "rock2"
    resource_name: "YKUSHPowerPort"
  }
  cls: "NetworkYKUSHPowerPort"
}

add resource pcduino3/USBSerialPort: USBSerialPort/OrderedDict([('match', OrderedDict([('@ID_PATH', 'pci-0000:03:00.2-usb-0:5.3.1:1.0')]))])
SerialPortExport(data={'avail': False, 'cls': 'NetworkSerialPort', 'params': OrderedDict(), 'acquired': None}, host='kea', proxy='kea', proxy_required=False, local=USBSerialPort(target=None, name=None, state=<BindingState.idle: 0>, avail=True, timeout=5.0, match=OrderedDict([('@ID_PATH', 'pci-0000:03:00.2-usb-0:5.3.1:1.0'), ('SUBSYSTEM', 'tty'), ('@SUBSYSTEM', 'usb')]), device=Device('/sys/devices/pci0000:00/0000:00:1b.0/0000:02:00.0/0000:03:00.2/usb1/1-5/1-5.3/1-5.3.1/1-5.3.1:1.0/ttyUSB14/tty/ttyUSB14'), port='/dev/ttyUSB14', speed=115200), local_params=OrderedDict([('match', OrderedDict([('@ID_PATH', 'pci-0000:03:00.2-usb-0:5.3.1:1.0'), ('SUBSYSTEM', 'tty'), ('@SUBSYSTEM', 'usb')]))]), start_params=None)
{'cls': 'NetworkSerialPort', 'params': OrderedDict(), 'acquired': None, 'avail': False}
encoded resource {
  path {
    group_name: "pcduino3"
    resource_name: "USBSerialPort"
  }
  cls: "NetworkSerialPort"
}

add resource pcduino3/USBSDWireDevice: USBSDWireDevice/OrderedDict([('match', OrderedDict([('ID_SERIAL_SHORT', 'sdwire-7')]))])
USBSDWireExport(data={'avail': False, 'cls': 'NetworkUSBSDWireDevice', 'params': OrderedDict(), 'acquired': None}, host='kea', proxy='kea', proxy_required=False, local=USBSDWireDevice(target=None, name=None, state=<BindingState.idle: 0>, avail=False, timeout=5.0, match=OrderedDict([('ID_SERIAL_SHORT', 'sdwire-7'), ('ID_VENDOR_ID', '04e8'), ('ID_MODEL_ID', '6001'), ('@ID_VENDOR_ID', '0424'), ('@ID_MODEL_ID', '2640'), ('SUBSYSTEM', 'usb')]), device=Device('/sys/devices/pci0000:00/0000:00:1b.0/0000:02:00.0/0000:03:00.2/usb1/1-5/1-5.4/1-5.4.2/1-5.4.2.2'), control_path=None, disk_path=None), local_params=OrderedDict([('match', OrderedDict([('ID_SERIAL_SHORT', 'sdwire-7'), ('ID_VENDOR_ID', '04e8'), ('ID_MODEL_ID', '6001'), ('@ID_VENDOR_ID', '0424'), ('@ID_MODEL_ID', '2640'), ('SUBSYSTEM', 'usb')]))]), start_params=None)
{'cls': 'NetworkUSBSDWireDevice', 'params': OrderedDict(), 'acquired': None, 'avail': False}
encoded resource {
  path {
    group_name: "pcduino3"
    resource_name: "USBSDWireDevice"
  }
  cls: "NetworkUSBSDWireDevice"
}

add resource pcduino3/YKUSHPowerPort: YKUSHPowerPort/OrderedDict([('serial', 'YK17698'), ('index', 2)])
YKUSHPowerPortExport(data={'avail': False, 'cls': 'NetworkYKUSHPowerPort', 'params': OrderedDict(), 'acquired': None}, host='kea', proxy='kea', proxy_required=False, local=YKUSHPowerPort(target=None, name=None, state=<BindingState.idle: 0>, avail=True, serial='YK17698', index=2), local_params=OrderedDict([('serial', 'YK17698'), ('index', 2)]), start_params=None)
{'cls': 'NetworkYKUSHPowerPort', 'params': OrderedDict(), 'acquired': None, 'avail': False}
encoded resource {
  path {
    group_name: "pcduino3"
    resource_name: "YKUSHPowerPort"
  }
  cls: "NetworkYKUSHPowerPort"
}

add resource pcduino3/SunxiUSBLoader: SunxiUSBLoader/OrderedDict([('match', OrderedDict([('ID_PATH', 'pci-0000:03:00.2-usb-0:5.1.2.2')]))])
ResourceEntry(data={'avail': True, 'cls': 'SunxiUSBLoader', 'params': OrderedDict([('match', OrderedDict([('ID_PATH', 'pci-0000:03:00.2-usb-0:5.1.2.2')])), ('extra', {'proxy': 'kea', 'proxy_required': False})]), 'acquired': None})
{'cls': 'SunxiUSBLoader', 'params': OrderedDict([('match', OrderedDict([('ID_PATH', 'pci-0000:03:00.2-usb-0:5.1.2.2')])), ('extra', {'proxy': 'kea', 'proxy_required': False})]), 'acquired': None, 'avail': True}
Traceback (most recent call last):
  File "/home/sglass/.local/bin/labgrid-exporter", line 8, in <module>
    sys.exit(main())
  File "/home/sglass/.local/lib/python3.10/site-packages/labgrid/remote/exporter.py", line 1014, in main
    asyncio.run(amain(config), debug=bool(args.debug))
  File "/usr/lib/python3.10/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.10/asyncio/base_events.py", line 649, in run_until_complete
    return future.result()
  File "/home/sglass/.local/lib/python3.10/site-packages/labgrid/remote/exporter.py", line 944, in amain
    await exporter.run()
  File "/home/sglass/.local/lib/python3.10/site-packages/labgrid/remote/exporter.py", line 777, in run
    await self.add_resource(
  File "/home/sglass/.local/lib/python3.10/site-packages/labgrid/remote/exporter.py", line 926, in add_resource
    await self.update_resource(group_name, resource_name)
  File "/home/sglass/.local/lib/python3.10/site-packages/labgrid/remote/exporter.py", line 935, in update_resource
    msg.resource.CopyFrom(resource.as_pb2())
  File "/home/sglass/.local/lib/python3.10/site-packages/labgrid/remote/common.py", line 121, in as_pb2
    set_map_from_dict(msg.params, params)
  File "/home/sglass/.local/lib/python3.10/site-packages/labgrid/remote/common.py", line 45, in set_map_from_dict
    raise ValueError(f"cannot translate {repr(v)} to MapValue")
ValueError: cannot translate OrderedDict([('ID_PATH', 'pci-0000:03:00.2-usb-0:5.1.2.2')]) to MapValue

```

sjg20, 2024-06-12:
[asc.txt](https://github.com/user-attachments/files/15805824/asc.txt)

jluebbe, 2024-06-12:
The match properties usually aren't part of the network exports... Could send your exporter configuration?

sjg20, 2024-06-12:
OK thank you. Actually I am not even using my new code. Presumably I would need to rebase on top of the grpc branch?

sjg20, 2024-06-12:
OK thanks.

I did a rebase, but I'm actually not sure how to run the exporter. Is there a docs update? I tried:

$ labgrid-coordinator -l kea:20408
INFO:root:loaded 34 place(s)

then

$ LG_COORDINATOR=kea:20408 labgrid-exporter ../cfg/export_kea.yaml 
exporter name: kea
exporter hostname: kea
resource config file: ../cfg/export_kea.yaml
INFO:root:creating poll task
Exporting completed (129 resources)
ERROR:root:coordinator is unavailable: DNS resolution failed for kea:20408: C-ares status is not ARES_SUCCESS qtype=A name=kea is_balancer=0: Misformatted domain name

What is the format for LG_COORDINATOR?

jluebbe, 2024-06-12:
There's still something broken with acquiring the resources on the exporter when locking places, which I'm looking at now. I'll keep you updated on IRC/Matrix.

sjg20, 2024-06-13:
I just tried the latest grpc branch and it works! So I am closing this.

closed 2024-06-13 by sjg20