labgrid-project/labgrid #1425: How to best add support for tp link switches?

colin-foster-in-advantage, 2024-06-14:
The TPLink power driver has support for power strips by way of the SmartStrip class. It uses the yaml `index` parameter to determine which child to turn on / off.

The HS103 is a single switch, so it isn't a SmartStrip with child devices. I'm looking to add support for the HS103, and I can imagine three routes:

1. Use a "magic index" of -1 to indicate "this is a SmartPlug, not a SmartStrip"
2. Add a new parameter in the yaml to indicate this
3. Discover the device type when the index is zero

I have a prototype of 1. I think 2 might be cleaner but I don't know how that would look.
3 I think might impact performance. I can poke around more if this might make sense.

Any thoughts / suggestions?

Emantor, 2024-09-09:
Does the smartplug use the same python module? If this is the case i'd make the driver compatible with both strips and plugs, leave the strips resource alone and add support for plugs as a new resource without an `index`. You can than modify the driver to only use the `index` for strips.

colin-foster-in-advantage, 2024-09-09:
I like the suggestion. My current status is I've bought a couple more strips and am not using a plug, so I probably won't get around to this one. If I do, I'll send a pull request!

closed 2024-09-09 by colin-foster-in-advantage