labgrid-project/labgrid #1429: Assume port 20408 for GRPC exporter

aparcar, 2024-06-24:
We assume port 20408 already for coordinator and client, however not for the exporter. I suggest we check if the line mentioned below checks for a port, and if it does not exists, add port 20408.

https://github.com/labgrid-project/labgrid/blob/ae0907928feec6b4bbaba7658cf5095d241a848d/labgrid/remote/exporter.py#L1028

Emantor, 2024-06-24:
~We do, we either get the env variable `LG_COORDINATOR` or fallback to `127.0.0.1:20408`:~
https://github.com/labgrid-project/labgrid/blob/ae0907928feec6b4bbaba7658cf5095d241a848d/labgrid/remote/exporter.py#L989

Ah, only the port argument is inconsistent. I guess @Bastian-Krause fixed this for the client since the client requires the proxy-manager to work, I am not against a fix for the exporter.

aparcar, 2024-06-26:
I guess this doesn't have a priority, it's just a inconsistency in the setup.

commit e3dbeccd (2024-08-08): remote/exporter: make coordinator port default to 20408

The coordinator binds to port 20408 by default. The client tries to
connect to that port if no port is specified via --cordinator/-x,
LG_COORDINATOR or via coordinator_address option in the environment
config.

Only the exporter does not yet default to 20408. Change that by
appending the default port 20408 to the address given via
--coordinator/-c or LG_COORDINATOR if no port was specified.

Closes #1429
commit 56d616a9 (2024-08-13): remote/exporter: make coordinator port default to 20408

The coordinator binds to port 20408 by default. The client tries to
connect to that port if no port is specified via --cordinator/-x,
LG_COORDINATOR or via coordinator_address option in the environment
config.

Only the exporter does not yet default to 20408. Change that by
appending the default port 20408 to the address given via
--coordinator/-c or LG_COORDINATOR if no port was specified.

Closes #1429
commit d7ac3e7f (2024-08-13): remote/exporter: make coordinator port default to 20408

The coordinator binds to port 20408 by default. The client tries to
connect to that port if no port is specified via --cordinator/-x,
LG_COORDINATOR or via coordinator_address option in the environment
config.

Only the exporter does not yet default to 20408. Change that by
appending the default port 20408 to the address given via
--coordinator/-c or LG_COORDINATOR if no port was specified.

Closes #1429
closed 2024-08-16 by Emantor