labgrid-project/labgrid #1457: Add DigitalInputProtocol, GpioDigitalInputDriver

flxzt, 2024-07-31:
Currently it is only possible to specify GPIO's as outputs [see here](https://github.com/labgrid-project/labgrid/blob/64b995fb7a6ef32fb40d15a54ee7dde5603520a4/labgrid/util/agents/sysfsgpio.py#L36).

I am proposing to add a `DigitalInputProtocol` together with a GpioDigitalInputDriver that configures GPIO pins as input only through sysfs (or switch to the newer devfs/libgpiod based GPIO interface while at it, because the sysfs one is deprecated).
This ensures that there won't be any shorts when the pin is connected to an output of the DUT.

flxzt, 2024-07-31:
I see. Having to keep a process open is a limitation of the devfs based interface that I didn't have in mind. Yes, I intend to add it. I'll see if I can come up with a sysfs based implementation similar to the existing agent code then

ozan956, 2026-10-30:
Since #1935 has been merged, we now have the DigitalInputProtocol and GpioDigitalInputDriver.

Thanks to everyone who contributed!

Closing this as resolved.

closed 2026-10-30 by ozan956