labgrid-project/labgrid #1459: Make `labgrid-client` work with local-only configuration

flxzt, 2024-08-01:
Currently it is only possible to use the labgrid-client tool with the distributed infrastructure - deployed exporter, coordinator, and so on.

When trying to use `labgrid-client` with only a local environment configuration where no remote place is specified and no coordinator address is supplied, the client errors out with message:

```
No RemotePlace found in configuration file
```

IMHO it would be nice to close the feature gap of the client for local usage by either use the locally defined resources directly for some of it's commands like `io`, or `console`, or start coordinator&exporter on-demand internally in some way.

luke-hackwell, 2025-01-28:
Hi, I've been experimenting to get `labgrid-client` to work without the coordinator deployed. I'd like to get some opinions of if this design could be useful.

You can see my changes in [this commit](https://github.com/ARM-software/labgrid/commit/baef33d53685e098cdee2217a2ee4afb40cd66a3). The diff looks large, but there aren't actually that many meaningful changes.

I've split the client into a remote and local part. The remote part works as the client does now. The local part can interact with devices, but can't do anything with places. 

The existing `ClientSession` class has been split into a base `ClientSession` and subclasses of `RemoteSession` and `LocalSession`.  The base `ClientSession` implements functions that interact with devices and other common functionality. `RemoteSession` implements functionality for places on top of this. `RemoteSession` and `LocalSession` implement the `_get_target` method, which allows the `LocalSession` to get targets without the need for places.

CL args have been split between remote and local. A new CL target `labgrid-client-local` has been made that contains only commands to run in local mode. In local mode the client will run based on the resources and drivers specified in the `main` target of the config file.

To try this:
```
labgrid-local-client -c ssh.yaml ssh
```
_ssh.yaml_
```
targets:
  main:
    resources:
      NetworkService:
        address: "localhost"
        username: "lukhac01"
    drivers:
      SSHDriver: {}
```

I've only tested the `ssh` command, but it seems like the others should also work (apart from the `console` command because it is async).

luke-hackwell, 2025-01-28:
I also tried spinning up a local coordinator. In this case you'd have to create places, add matches, export resources and such behind the scenes, which seems like more effort and a worse design than the other approach

luke-hackwell, 2025-04-03:
Hi @Emantor, are there any thoughts on this?

Emantor, 2025-04-08:
I don't think using a spearate command for the local client is strictly necessary, We should be able to detect whether a `ClientSession` is required at all by looking into the configuration file early and figuring out if it contains a `RemotePlace` resource. If it does not we should be able to operate without a ClientSession.