labgrid-project/labgrid #1468: v24.0 StepLogger may be started twice which results in errors.

jremmet, 2024-08-13:
We use [pytester](https://docs.pytest.org/en/stable/reference/reference.html#pytester) to test our hooks together with labgrid.
Since v24.0 we run into errors because `StepLogger.start` is called again by  pytester via `pytest_configure` hook.

minimal example:
test_mini.py
```
import pytest

pytest_plugins = ["pytester"]

DUMMY_TEST = """
import pytest

def test_success():
    pass
"""

def test_pytester(pytester):
    """Test pytester."""
    pytester.makepyfile(DUMMY_TEST)
    result = pytester.runpytest()
    assert result.ret == pytest.ExitCode.OK
```

```
==================================================================== test session starts ====================================================================
platform linux -- Python 3.10.12, pytest-8.3.2, pluggy-1.5.0
rootdir: /tmp/labgtid
plugins: labgrid-24.0
collected 1 item                                                                                                                                            

test_mini.py F                                                                                                                                        [100%]

========================================================================= FAILURES ==========================================================================
_______________________________________________________________________ test_pytester _______________________________________________________________________

pytester = <Pytester PosixPath('/tmp/pytest-of-jremmet/pytest-0/test_pytester0')>

    def test_pytester(pytester):
        """Test pytester."""
        pytester.makepyfile(DUMMY_TEST)
        result = pytester.runpytest()
>       assert result.ret == pytest.ExitCode.OK
E       AssertionError: assert <ExitCode.INTERNAL_ERROR: 3> == <ExitCode.OK: 0>
E        +  where <ExitCode.INTERNAL_ERROR: 3> = <RunResult ret=ExitCode.INTERNAL_ERROR len(stdout.lines)=19 len(stderr.lines)=0 duration=0.02s>.ret
E        +  and   <ExitCode.OK: 0> = <enum 'ExitCode'>.OK
E        +    where <enum 'ExitCode'> = pytest.ExitCode

/tmp/labgtid/test_mini.py:16: AssertionError
------------------------------------------------------------------- Captured stdout call --------------------------------------------------------------------
INTERNALERROR> Traceback (most recent call last):
INTERNALERROR>   File "/tmp/labgtid/venv/lib/python3.10/site-packages/_pytest/main.py", line 279, in wrap_session
INTERNALERROR>     config._do_configure()
INTERNALERROR>   File "/tmp/labgtid/venv/lib/python3.10/site-packages/_pytest/config/__init__.py", line 1118, in _do_configure
INTERNALERROR>     self.hook.pytest_configure.call_historic(kwargs=dict(config=self))
INTERNALERROR>   File "/tmp/labgtid/venv/lib/python3.10/site-packages/pluggy/_hooks.py", line 535, in call_historic
INTERNALERROR>     res = self._hookexec(self.name, self._hookimpls.copy(), kwargs, False)
INTERNALERROR>   File "/tmp/labgtid/venv/lib/python3.10/site-packages/pluggy/_manager.py", line 120, in _hookexec
INTERNALERROR>     return self._inner_hookexec(hook_name, methods, kwargs, firstresult)
INTERNALERROR>   File "/tmp/labgtid/venv/lib/python3.10/site-packages/pluggy/_callers.py", line 139, in _multicall
INTERNALERROR>     raise exception.with_traceback(exception.__traceback__)
INTERNALERROR>   File "/tmp/labgtid/venv/lib/python3.10/site-packages/pluggy/_callers.py", line 103, in _multicall
INTERNALERROR>     res = hook_impl.function(*args)
INTERNALERROR>   File "/tmp/labgtid/venv/lib/python3.10/site-packages/labgrid/pytestplugin/hooks.py", line 65, in pytest_configure
INTERNALERROR>     StepLogger.start()
INTERNALERROR>   File "/tmp/labgtid/venv/lib/python3.10/site-packages/labgrid/logging.py", line 147, in start
INTERNALERROR>     assert not cls._started
INTERNALERROR> AssertionError: assert not True
INTERNALERROR>  +  where True = <class 'labgrid.logging.StepLogger'>._started
================================================================== short test summary info ==================================================================
FAILED test_mini.py::test_pytester - AssertionError: assert <ExitCode.INTERNAL_ERROR: 3> == <ExitCode.OK: 0>
===================================================================== 1 failed in 0.05s =====================================================================
```

Is it possible to replace the asserion by a if clause here:
https://github.com/labgrid-project/labgrid/blob/7f2c8fc95e22a64a310e9ad8d3ce56d614e7e2f1/labgrid/logging.py#L147
Or add a "is-already-running" check here:
https://github.com/labgrid-project/labgrid/blob/7f2c8fc95e22a64a310e9ad8d3ce56d614e7e2f1/labgrid/pytestplugin/hooks.py#L65
?

Emantor, 2024-08-13:
Thats unfortunate. I guess allowing multiple starts for the `StepLogger` should not be too bad, so I think we should handle that in the the `Steplogger` class, if @jluebbe and @Bastian-Krause agree.

jremmet, 2024-08-13:
> Thats unfortunate. I guess allowing multiple starts for the `StepLogger` should not be too bad, so I think we should handle that in the the `Steplogger` class, if @jluebbe and @Bastian-Krause agree.

I added a PR which handle it in the pytest hook. If we just ignore multiple calls, we will run `stop` to early.
Alternatively we need to count the starts and stops, which seams fragile if called directly.

Bastian-Krause, 2024-08-15:
> Thats unfortunate. I guess allowing multiple starts for the `StepLogger` should not be too bad, so I think we should handle that in the the `Steplogger` class, if @jluebbe and @Bastian-Krause agree.

I don't know the implications of that. As long as you say it's safe, I have no objections.

jremmet, 2024-11-07:
gently ping