labgrid-project/labgrid #1476: Unable to build the Docker container from the instructions

graugans, 2024-08-15:
What I did:

```sh
git clone https://github.com/labgrid-project/labgrid.git
cd labgrid/
git checkout -b v24.0 v24.0
docker build --target labgrid-coordinator -t nexus3.dev.ifm:18443/labgrid-coordinator:24.0 -f dockerfiles/Dockerfile .
```

This is the error I do receive:
```
------
 > [labgrid-coordinator 1/1] RUN set -e ;    cd /opt/labgrid ;    pip3 install --break-system-packages virtualenv ;    SETUPTOOLS_SCM_PRETEND_VERSION="$VERSION" pip3 install --break-system-packages --no-cache-dir . ;    virtualenv -p python3 crossbar-venv ;    crossbar-venv/bin/pip3 install --break-system-packages -r crossbar-requirements.txt ;    sed -i "s#^  executable: .*$#  executable: python3#" .crossbar/config-anonymous.yaml:
1.112 Collecting virtualenv
1.455   Downloading virtualenv-20.26.3-py3-none-any.whl.metadata (4.5 kB)
1.563 Collecting distlib<1,>=0.3.7 (from virtualenv)
1.619   Downloading distlib-0.3.8-py2.py3-none-any.whl.metadata (5.1 kB)
1.800 Collecting filelock<4,>=3.12.2 (from virtualenv)
1.853   Downloading filelock-3.15.4-py3-none-any.whl.metadata (2.9 kB)
1.986 Collecting platformdirs<5,>=3.9.1 (from virtualenv)
2.041   Downloading platformdirs-4.2.2-py3-none-any.whl.metadata (11 kB)
2.152 Downloading virtualenv-20.26.3-py3-none-any.whl (5.7 MB)
2.502    ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 5.7/5.7 MB 19.9 MB/s eta 0:00:00
2.557 Downloading distlib-0.3.8-py2.py3-none-any.whl (468 kB)
2.639 Downloading filelock-3.15.4-py3-none-any.whl (16 kB)
2.720 Downloading platformdirs-4.2.2-py3-none-any.whl (18 kB)
2.799 Installing collected packages: distlib, platformdirs, filelock, virtualenv
3.102 Successfully installed distlib-0.3.8 filelock-3.15.4 platformdirs-4.2.2 virtualenv-20.26.3
3.103 WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager, possibly rendering your system unusable.It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv. Use the --root-user-action option if you know what you are doing and want to suppress this warning.
3.696 Processing /opt/labgrid
3.699   Installing build dependencies: started
6.742   Installing build dependencies: finished with status 'done'
6.743   Getting requirements to build wheel: started
6.848   Getting requirements to build wheel: finished with status 'error'
6.852   error: subprocess-exited-with-error
6.852
6.852   × Getting requirements to build wheel did not run successfully.
6.852   │ exit code: 1
6.852   ╰─> [40 lines of output]
6.852       Traceback (most recent call last):
6.852         File "/usr/local/lib/python3.11/dist-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 353, in <module>
6.852           main()
6.852         File "/usr/local/lib/python3.11/dist-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 335, in main
6.852           json_out['return_val'] = hook(**hook_input['kwargs'])
6.852                                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
6.852         File "/usr/local/lib/python3.11/dist-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 118, in get_requires_for_build_wheel
6.852           return hook(config_settings)
6.852                  ^^^^^^^^^^^^^^^^^^^^^
6.852         File "/tmp/pip-build-env-tygcu3v7/overlay/local/lib/python3.11/dist-packages/setuptools/build_meta.py", line 327, in get_requires_for_build_wheel
6.852           return self._get_build_requires(config_settings, requirements=[])
6.852                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
6.852         File "/tmp/pip-build-env-tygcu3v7/overlay/local/lib/python3.11/dist-packages/setuptools/build_meta.py", line 297, in _get_build_requires
6.852           self.run_setup()
6.852         File "/tmp/pip-build-env-tygcu3v7/overlay/local/lib/python3.11/dist-packages/setuptools/build_meta.py", line 313, in run_setup
6.852           exec(code, locals())
6.852         File "<string>", line 1, in <module>
6.852         File "/tmp/pip-build-env-tygcu3v7/overlay/local/lib/python3.11/dist-packages/setuptools/__init__.py", line 108, in setup
6.852           return distutils.core.setup(**attrs)
6.852                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
6.852         File "/tmp/pip-build-env-tygcu3v7/overlay/local/lib/python3.11/dist-packages/setuptools/_distutils/core.py", line 146, in setup
6.852           _setup_distribution = dist = klass(attrs)
6.852                                        ^^^^^^^^^^^^
6.852         File "/tmp/pip-build-env-tygcu3v7/overlay/local/lib/python3.11/dist-packages/setuptools/dist.py", line 278, in __init__
6.852           _Distribution.__init__(self, dist_attrs)
6.852         File "/tmp/pip-build-env-tygcu3v7/overlay/local/lib/python3.11/dist-packages/setuptools/_distutils/dist.py", line 286, in __init__
6.852           self.finalize_options()
6.852         File "/tmp/pip-build-env-tygcu3v7/overlay/local/lib/python3.11/dist-packages/setuptools/dist.py", line 630, in finalize_options
6.852           ep(self)
6.852         File "/tmp/pip-build-env-tygcu3v7/overlay/local/lib/python3.11/dist-packages/setuptools_scm/_integration/setuptools.py", line 123, in infer_version
6.852           _assign_version(dist, config)
6.852         File "/tmp/pip-build-env-tygcu3v7/overlay/local/lib/python3.11/dist-packages/setuptools_scm/_integration/setuptools.py", line 58, in _assign_version
6.852           _version_missing(config)
6.852         File "/tmp/pip-build-env-tygcu3v7/overlay/local/lib/python3.11/dist-packages/setuptools_scm/_get_version_impl.py", line 117, in _version_missing
6.852           raise LookupError(
6.852       LookupError: setuptools-scm was unable to detect version for /opt/labgrid.
6.852
6.852       Make sure you're either building from a fully intact git repository or PyPI tarballs. Most other sources (such as GitHub's tarballs, a git checkout without the .git folder) don't contain the necessary metadata and will not work.
6.852
6.852       For example, if you're using pip, instead of https://github.com/user/proj/archive/master.zip use git+https://github.com/user/proj.git#egg=proj
6.852       [end of output]
6.852
6.852   note: This error originates from a subprocess, and is likely not a problem with pip.
7.287 error: subprocess-exited-with-error
7.287
7.287 × Getting requirements to build wheel did not run successfully.
7.287 │ exit code: 1
7.287 ╰─> See above for output.
7.287
7.287 note: This error originates from a subprocess, and is likely not a problem with pip.
------
Dockerfile:41
--------------------
  40 |
  41 | >>> RUN set -e ;\
  42 | >>>     cd /opt/labgrid ;\
  43 | >>>     pip3 install --break-system-packages virtualenv ;\
  44 | >>>     SETUPTOOLS_SCM_PRETEND_VERSION="$VERSION" pip3 install --break-system-packages --no-cache-dir . ;\
  45 | >>>     virtualenv -p python3 crossbar-venv ;\
  46 | >>>     crossbar-venv/bin/pip3 install --break-system-packages -r crossbar-requirements.txt ;\
  47 | >>>     sed -i "s#^  executable: .*\$#  executable: python3#" .crossbar/config-anonymous.yaml
  48 |
--------------------
ERROR: failed to solve: process "/bin/sh -c set -e ;    cd /opt/labgrid ;    pip3 install --break-system-packages virtualenv ;    SETUPTOOLS_SCM_PRETEND_VERSION=\"$VERSION\" pip3 install --break-system-packages --no-cache-dir . ;    virtualenv -p python3 crossbar-venv ;    crossbar-venv/bin/pip3 install --break-system-packages -r crossbar-requirements.txt ;    sed -i \"s#^  executable: .*\\$#  executable: python3#\" .crossbar/config-anonymous.yaml" did not complete successfully: exit code: 1
```

I am not sure if this is related to: https://github.com/pypa/setuptools_scm/issues/1011

Bastian-Krause, 2024-08-15:
Looking at..
https://github.com/labgrid-project/labgrid/blob/f472ae72ab8f5969c15c9d2af8f22deeeb00bb85/dockerfiles/build.sh#L63-L65
..you probably need `--build-arg VERSION="$(python -m setuptools_scm)"` in your docker build command. If it works, please add it to the README.

graugans, 2024-08-16:
Okay, that was pointing me in the right direction. Maybe it is my old Ubuntu 20.04 or my specific Python set-up. This worked for me:

```sh
 python3 -m venv .venv
. .venv/bin/activate
pip install --upgrade pip setuptools setuptools_scm
docker build --target labgrid-coordinator  --build-arg VERSION="$(python -m setuptools_scm)"  -t nexus3.dev.ifm:18443/labgrid-coordinator:24.0 -f dockerfiles/Dockerfile .
```

If the virtualenv is fine for you I'll create an MR for the Readme.

Without the `virtualenv` I do get:

```sh
pip3 install --upgrade setuptools setuptools_scm
python3 -m setuptools_scm
/usr/bin/python3: No module named setuptools_scm
```

When I just install `setuptools_scm` I do get this warning/error:

```
ERROR: setuptools==45.2.0 is used in combination with setuptools_scm>=8.x                                                                                                                
                                                                                                                                                                                         
Your build configuration is incomplete and previously worked by accident!                                                                                                                
setuptools_scm requires setuptools>=61                                                                                                                                                   
                                                                                                                                                                                         
Suggested workaround if applicable:                                                                                                                                                      
 - migrating from the deprecated setup_requires mechanism to pep517/518                                                                                                                  
   and using a pyproject.toml to declare build dependencies                                                                                                                              
   which are reliably pre-installed before running the build tools                                                                                                                       
                                                                                                                                                                                         
  warnings.warn(                                                                                                                                                                         
24.0
```

Emantor, 2024-09-09:
I guess upgrading these dependencies inside the virtualenv is the only way. Note that `python3.8` in ubuntu focal will eventually be dropped from labgrid support, since python 3.8 support will end in october 2024.