labgrid-project/labgrid #1500: ssh user

josepee, 2024-09-26:
Hello, 

I'm building a Labgrid infrastructure and have a problem with commands using ssh.
It happens when i use the UBSDMux or the SiSPMPower controller.
When i run this command:
```
$ labgrid-client -v -d -p mx6 power on
```
i get this output:
```
DEBUG              root: Starting session with "ws://10.1.75.68:20408/ws", realm: "realm1"
DEBUG   RemotePlaceMana: expanded remote resources for place mx6: [NetworkUSBSDMuxDevice(target=Target(name='mx6', env=None), name='USBSDMuxDevice', state=<BindingState.bound: 1>, avail=True, host='10.1.75.68', timeout=10.0, busnum=1, devnum=5, path='/dev/sda', vendor_id=1060, model_id=16449, control_path='/dev/sg0'), NetworkSerialPort(target=Target(name='mx6', env=None), name='USBSerialPort', state=<BindingState.bound: 1>, avail=True, host='10.1.75.68', port=46383, speed=115200, protocol='rfc2217'), NetworkSiSPMPowerPort(target=Target(name='mx6', env=None), name='SiSPMPowerPort', state=<BindingState.bound: 1>, avail=True, host='10.1.75.68', timeout=10.0, busnum=1, devnum=6, path='1-1.4', vendor_id=1204, model_id=64789, index=1), NetworkService(target=Target(name='mx6', env=None), name='NetworkService', state=<BindingState.bound: 1>, avail=True, address='10.1.75.78', username='root', password='', port=22)]
INFO         StepLogger:  → SiSPMPowerDriver.on() 
DEBUG   SSHConnectionMa:   Creating SSHConnection for 10.1.75.68
DEBUG   SSHConnection(h:   No existing control socket found
DEBUG   SSHConnection(h:   ControlSocket: /tmp/lg-con-5m2dy9iz/control-10.1.75.68
DEBUG   SSHConnection(h:   Master Start command: ssh -x -o LogLevel=ERROR -o PasswordAuthentication=no -n -MN -o ConnectTimeout=30 -o ControlPersist=300 -o ControlMaster=yes -o ControlPath=/tmp/lg-con-5m2dy9iz/control-10.1.75.68 -o StrictHostKeyChecking=yes 10.1.75.68
INFO         StepLogger:  ⚠ SiSPMPowerDriver.on() [3.228s] exception=failed to connect to 10.1.75.68 with args ['ssh', '-x', '-o', 'LogLevel=ERROR', '-o', 'PasswordAuthentication=no', '-n', '-MN', '-o', 'ConnectTimeout=30', '-o', 'ControlPersist=300', '-o', 'ControlMaster=yes', '-o', 'ControlPath=/tmp/lg-con-5m2dy9iz/control-10.1.75.68', '-o', 'StrictHostKeyChecking=yes', '10.1.75.68'], returncode=255 b'',b'juepe@10.1.75.68: Permission denied (publickey,password).\r\n'
Traceback (most recent call last):
  File "/home/juepe/.local/lib/python3.10/site-packages/labgrid/remote/client.py", line 1940, in main
    args.func(session)
  File "/home/juepe/.local/lib/python3.10/site-packages/labgrid/remote/client.py", line 758, in power
    res = getattr(drv, action)()
  File "/home/juepe/.local/lib/python3.10/site-packages/labgrid/binding.py", line 102, in wrapper
    return func(self, *_args, **_kwargs)
  File "/home/juepe/.local/lib/python3.10/site-packages/labgrid/step.py", line 215, in wrapper
    _result = func(*_args, **_kwargs)
  File "/home/juepe/.local/lib/python3.10/site-packages/labgrid/driver/powerdriver.py", line 86, in on
    processwrapper.check_output(self._get_sispmctl_prefix() + cmd)
  File "/home/juepe/.local/lib/python3.10/site-packages/labgrid/driver/powerdriver.py", line 77, in _get_sispmctl_prefix
    return self.port.command_prefix + [
  File "/home/juepe/.local/lib/python3.10/site-packages/labgrid/resource/common.py", line 100, in command_prefix
    conn = sshmanager.get(host)
  File "/home/juepe/.local/lib/python3.10/site-packages/labgrid/util/ssh.py", line 50, in get
    instance.connect()
  File "/home/juepe/.local/lib/python3.10/site-packages/labgrid/util/ssh.py", line 400, in connect
    self._open_connection()
  File "/home/juepe/.local/lib/python3.10/site-packages/labgrid/util/ssh.py", line 174, in _open_connection
    self._start_own_master()
  File "/home/juepe/.local/lib/python3.10/site-packages/labgrid/util/ssh.py", line 458, in _start_own_master
    raise ExecutionError(
labgrid.driver.exception.ExecutionError: failed to connect to 10.1.75.68 with args ['ssh', '-x', '-o', 'LogLevel=ERROR', '-o', 'PasswordAuthentication=no', '-n', '-MN', '-o', 'ConnectTimeout=30', '-o', 'ControlPersist=300', '-o', 'ControlMaster=yes', '-o', 'ControlPath=/tmp/lg-con-5m2dy9iz/control-10.1.75.68', '-o', 'StrictHostKeyChecking=yes', '10.1.75.68'], returncode=255 b'',b'juepe@10.1.75.68: Permission denied (publickey,password).\r\n'
```
At the end it tells me that it tries to connect with the user juepe that is mine on my host but i set up a different username on the exporter.

I made a modification on the source code to understand the error, in labgrid/util/ssh.py:
BEFORE
```
def _start_own_master(self):
        """Starts a controlmaster connection in a temporary directory."""
        control = os.path.join(self._tmpdir, f'control-{self.host}')

        connect_timeout = get_ssh_connect_timeout()

        self._logger.debug("ControlSocket: %s", control)
        args = ["ssh"] + SSHConnection._get_ssh_base_args()
        args += [
            "-n", "-MN",
            "-o", f"ConnectTimeout={connect_timeout}",
            "-o", "ControlPersist=300",
            "-o", "ControlMaster=yes",
            "-o", f"ControlPath={control}",
            # We don't want to ask the user to confirm host keys here.
            "-o", "StrictHostKeyChecking=yes",
            self.host,
        ]
```
AFTER
```
def _start_own_master(self):
        """Starts a controlmaster connection in a temporary directory."""
        control = os.path.join(self._tmpdir, f'control-{self.host}')

        connect_timeout = get_ssh_connect_timeout()

        self._logger.debug("ControlSocket: %s", control)
        args = ["ssh"] + SSHConnection._get_ssh_base_args()
        args += [
            "-n", "-MN",
            "-o", f"ConnectTimeout={connect_timeout}",
            "-o", "ControlPersist=300",
            "-o", "ControlMaster=yes",
            "-o", f"ControlPath={control}",
            # We don't want to ask the user to confirm host keys here.
            "-o", "StrictHostKeyChecking=yes",
            "labgrid@" + self.host,
        ]
```
It now works, the users are the same at both ends.
Is there another way i'm missing to solve this out, another place i can specify the user and the host i want to connect to ?

Thank you in advance.

Emantor, 2024-09-26:
You'll want to setup an SSH config entry for this host which sets the user to the user you have configured on the device.

josepee, 2024-09-26:
You're a prince thank you!
I tried with ssh config used it wrongly now it works