labgrid-project/labgrid #1503: ssh options typo ?

josepee, 2024-09-27:
Hi again,

looking into the code of labgrid/util/ssh.py, in _start_own_master,  in the ssh otpion arguments, i see this:
```
# We don't want to ask the user to confirm host keys here.
"-o", "StrictHostKeyChecking=yes",
```
is it a typo and you meant no ?
Or just meant that there shouldn't be a prompt to fill in but the key should be known ?

Emantor, 2024-09-27:
What the comment means is that we want to user to have already logged into the machine, otherwise we error out. This is necessary since we can't handle the confirmation prompt within labgrid and want the code to return an error in this case.

josepee, 2024-09-27:
ok so i have to connect once by hand to the ssh server and then i can use it freely

Emantor, 2024-09-27:
Yes, this is the idea.

josepee, 2024-09-27:
ok thank you :pray: :

closed 2024-09-27 by Emantor