labgrid-project/labgrid #1507: siglent power backend incompatible with Python 3.13

Bastian-Krause, 2024-09-30:
The siglent power backend uses the [vxi11](https://pypi.org/project/python-vxi11/) module. It hasn't seen any activity in the last 7 years. It still uses the deprecated [xdrlib](https://docs.python.org/3/library/xdrlib.html) module from the Python standard library. xdrlib will be removed in Python 3.13 which is [expected next week](https://peps.python.org/pep-0719/#schedule).

Upstream issue: https://github.com/python-ivi/python-vxi11/issues/49

If nobody steps in, we will deprecate the siglent power backend and drop it in the foreseeable future.

Cc @esben (original contributor of the siglent power backend)

heevasti, 2024-10-23:
Hi Bastian, we had a similar problem with our Tektronix 5014 AWG. Our control software for it depends also on the use of `python-vxi11`. But, you can extend life by putting in project dependencies for Python 3.13 version to https://github.com/da4089/py-xdrlib. It install the `py-xdrlib` in the same namespace as the `xdrlib`, so the import statement won't raise an exception and will work 'as usual'. Note that the currently latest version in Pypi is still 4.0.1. which does not work. So the pip install needs (at the moment) be done directly from the Github project.

I hope this information helps, with that these old hardware can be made run a little longer with the old dependencies.

esben, 2024-12-11:
You could also use https://github.com/youknowone/python-deadlib, and install `standard-xdrlib` which should provide a clean backport of the removed xdrlib module.