labgrid-project/labgrid #1514: Multiple SysfsGPIO resources in one place/group?

nicolas-andersen, 2024-10-03:
I was under the impression that you could name resources like this in the exporter.yaml
`board1:
  RawSerialPort:
    port: '/dev/ttyUSB0'
    speed: 115200
  SysfsGPIO:
    index: 529
    name: 'gpio17'
  SysfsGPIO:
    index: 539
    name: 'gpio27'`
when I try to run labgrid-exporter, I get this error
`exporter name: test-pi
exporter hostname: test-pi
resource config file: exporter.yaml
('rendered',
 'board1:\n'
 '  RawSerialPort:\n'
 "    port: '/dev/ttyUSB0'\n"
 '    speed: 115200\n'
 '  SysfsGPIO:\n'
 '    index: 529\n'
 "    name: 'gpio17'\n"
 '  SysfsGPIO:\n'
 '    index: 539\n'
 "    name: 'gpio27'")
/opt/labgrid/.venv/lib/python3.11/site-packages/labgrid/util/yaml.py:26: UserWarning: <unicode string>: previous entry with duplicate YAML dictionary key 'SysfsGPIO' overwritten
  warnings.warn(
('loaded',
 OrderedDict([('board1',
               OrderedDict([('RawSerialPort',
                             OrderedDict([('port', '/dev/ttyUSB0'),
                                          ('speed', 115200)])),
                            ('SysfsGPIO',
                             OrderedDict([('index', 539),
                                          ('name', 'gpio27')]))]))]))
add resource board1/RawSerialPort: RawSerialPort/OrderedDict([('port', '/dev/ttyUSB0'), ('speed', 115200)])
INFO:root:queued update for resource board1/RawSerialPort
add resource board1/SysfsGPIO: SysfsGPIO/OrderedDict([('index', 539), ('name', 'gpio27')])
Traceback (most recent call last):
  File "/opt/labgrid/.venv/bin/labgrid-exporter", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/opt/labgrid/.venv/lib/python3.11/site-packages/labgrid/remote/exporter.py", line 1067, in main
    asyncio.run(amain(config), debug=bool(args.debug))
  File "/usr/lib/python3.11/asyncio/runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/opt/labgrid/.venv/lib/python3.11/site-packages/labgrid/remote/exporter.py", line 1007, in amain
    await exporter.run()
  File "/opt/labgrid/.venv/lib/python3.11/site-packages/labgrid/remote/exporter.py", line 835, in run
    await self.add_resource(group_name, resource_name, cls, params)
  File "/opt/labgrid/.venv/lib/python3.11/site-packages/labgrid/remote/exporter.py", line 985, in add_resource
    group[resource_name] = export_cls(config, host=self.hostname, proxy=getfqdn(), proxy_required=proxy_req)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<attrs generated init labgrid.remote.exporter.GPIOSysFSExport>", line 8, in __init__
  File "/opt/labgrid/.venv/lib/python3.11/site-packages/labgrid/remote/exporter.py", line 622, in __attrs_post_init__
    self.local = SysfsGPIO(target=None, name=None, **self.local_params)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: labgrid.resource.base.SysfsGPIO() got multiple values for keyword argument 'name'`

What is the proper way of dealing with multiple resources of the same type?

Bastian-Krause, 2024-10-15:
The usual way of adding multiple resources of the same class to a place is by using named matches, i.e. `labgrid-client add-named-match PATTERN NAME`

istepic, 2024-12-23:
I think what OP wanted to get is what is described in https://labgrid.readthedocs.io/en/latest/configuration.html#exporter-configuration

From the docs:
`usb-hub-in-rack12:
  console-main:
    cls: 'USBSerialPort'
    match:
      '@ID_PATH': 'pci-0000:05:00.0-usb-3-1.3'
  console-secondary:
    cls: 'USBSerialPort'
    match:
      '@ID_PATH': 'pci-0000:05:00.0-usb-3-1.4'
`

ozan956, 2026-10-13:
Since the issue appears to have been resolved through the discussion above and is related to configuration rather than a bug in labgrid itself, I will go ahead and close this issue.

For reference, the resource configuration should look something like this:

```yml
board1:
  RawSerialPort:
    port: '/dev/ttyUSB0'
    speed: 115200
  gpio17:
    cls: 'SysfsGPIO'
    index: 529
  gpio27:
    cls: 'SysfsGPIO'
    index: 539
```

Feel free to comment if anything remains unclear or if you believe the issue should be reopened.

Thanks!

closed 2026-10-13 by ozan956