labgrid-project/labgrid #1544: [labgrid-client] some successful commands return empty exceptions

QSchulz, 2024-11-13:
but they still do what they are supposed to do.

```console
$ labgrid-client power off
Selected role main from configuration file
Error in sys.excepthook:

Original exception was:
```

```console
$ labgrid-client io high
Selected role main from configuration file
Error in sys.excepthook:

Original exception was:
```

```console
$ labgrid-client sd-mux host
Selected role main from configuration file
Error in sys.excepthook:

Original exception was:
```

labgrid 0304ec6c Merge pull request #1525 from Bastian-Krause/bst/rawnet-conf
fedora41 with Python 3.13, labgrid within a venv without system packages I believe

```yaml
---
Ringneck-Haikou_01:
  USBSerialPort:
    match:
      ID_SERIAL_SHORT: 'e06508c3d885e811933eefa9885bad01'
      ID_USB_INTERFACE_NUM: '00'
    speed: 115200
  RKUSBLoader:
    match:
      ID_PATH: 'pci-0000:0a:00.4-usb-0:1.3'
  USBSDWireDevice:
    match:
      ID_PATH: 'pci-0000:0a:00.4-usb-0:1.2.2'
  CP2102BIOS:
    cls: 'MatchedSysfsGPIO'
    match:
      '@SUBSYSTEM': 'usb'
      '@ID_PATH': 'pci-0000:0a:00.4-usb-0:1.4:1.0'
    pin: 1
```
```yaml
Ringneck-Haikou_01:
  acquired: fedora/qschulz
  acquired_resources:
  - - desq1.lan
    - Ringneck-Haikou_01
    - NetworkSysfsGPIO
    - CP2102BIOS
  - - desq1.lan
    - Ringneck-Haikou_01
    - NetworkRKUSBLoader
    - RKUSBLoader
  - - desq1.lan
    - Ringneck-Haikou_01
    - NetworkUSBSDWireDevice
    - USBSDWireDevice
  - - desq1.lan
    - Ringneck-Haikou_01
    - NetworkSerialPort
    - USBSerialPort
  aliases: []
  allowed: []
  changed: 1731429538.0324962
  comment: ''
  created: 1731334662.555931
  matches:
  - cls: '*'
    exporter: '*'
    group: Ringneck-Haikou_01
    name: null
    rename: null
  reservation: null
  tags: {}
```
```yaml
desq1.lan:
  Ringneck-Haikou_01:
    CP2102BIOS:
      acquired: Ringneck-Haikou_01
      avail: true
      cls: NetworkSysfsGPIO
      params:
        extra:
          proxy: desq1.lan
          proxy_required: false
        host: desq1.lan
        index: 757
    RKUSBLoader:
      acquired: Ringneck-Haikou_01
      avail: false
      cls: NetworkRKUSBLoader
      params:
        busnum: null
        devnum: null
        extra:
          proxy: desq1.lan
          proxy_required: false
        host: desq1.lan
        model_id: null
        path: null
        vendor_id: null
    USBSDWireDevice:
      acquired: Ringneck-Haikou_01
      avail: true
      cls: NetworkUSBSDWireDevice
      params:
        busnum: 6
        control_serial: bdgrd_sdwirec_196
        devnum: 19
        extra:
          proxy: desq1.lan
          proxy_required: false
        host: desq1.lan
        model_id: 24577
        path: /dev/sda
        vendor_id: 1256
    USBSerialPort:
      acquired: Ringneck-Haikou_01
      avail: true
      cls: NetworkSerialPort
      params:
        extra:
          path: /dev/ttyUSB0
          proxy: desq1.lan
          proxy_required: false
        host: desq1.lan
        port: 42139
        speed: 115200
```
```yaml
---
targets:
  main:
    resources:
      RemotePlace:
        name: !template '$LG_PLACE'
    drivers:
      ExternalPowerDriver:
        cmd_on: 'ssh desq1.lan -- /home/theo/work/labgrid/venv/bin/python3 /home/theo/work/som-tools/usb-control/haikou.py --serial e06508c3d885e811933eefa9885bad01 on'
        cmd_off: 'ssh desq1.lan -- /home/theo/work/labgrid/venv/bin/python3 /home/theo/work/som-tools/usb-control/haikou.py --serial e06508c3d885e811933eefa9885bad01 off'
        cmd_cycle: 'ssh desq1.lan -- /home/theo/work/labgrid/venv/bin/python3 /home/theo/work/som-tools/usb-control/haikou.py --serial e06508c3d885e811933eefa9885bad01 cycle'
      GpioDigitalOutputDriver: {}
      RKUSBDriver:
        usb_loader: 'rkbin-db-px30'
      SerialDriver: {}
      ShellDriver:
        prompt: '\x1b\[\?2004hroot@[\w-]+:[^ ]+ '
        login_prompt: ' login: '
        username: 'root'
        password: 'root'
        post_login_settle_time: 2
      UBootDriver:
        prompt: '=> '
      USBStorageDriver: {}
      USBSDWireDriver: {}

      HaikouStrategy: {}
imports:
  - haikoustrategy.py
images:
  sd_uboot: "u-boot-rockchip.bin.ringneck"
  rkbin-db-px30: "rkbin-db-px30.bin"

tools:
  rk-usb-loader: 'rkdeveloptool'
```

Bastian-Krause, 2024-11-13:
I've seen similar messages with Python 3.13. I haven't found the time to debug this. Help appreciated.

Bastian-Krause, 2024-11-13:
Might be related to https://github.com/grpc/grpc/issues/36655 and 
https://github.com/grpc/grpc/issues/33342

If I remember correctly, I saw this only with Python3.13. But the issues linked above report this also on other versions. Needs investigation.

pmelange, 2025-01-27:
I am also seeing this when using labgrid-client console.  The error happens after using the Escape Character Ctrl-\ and then typing "quit".  

Commit ID in use: f81f47975434784e9a5919b5fd369bdcea783462

Bastian-Krause, 2025-02-12:
We've merged #1605 as a work around until it's fixed in gRPC.

Emantor, 2026-10-10:
Closing since we have a workaround and won't bump gRPC for a longer time.

closed 2026-10-10 by Emantor