labgrid-project/labgrid #1585: docker: build.sh: has_buildx returns false positive

sgross-emlix, 2025-01-29:
# Reproduction
```
dockerfiles/build.sh
```

# Observed Behaviour (with -x)

```
+ export DOCKER_BUILDKIT=1
+ DOCKER_BUILDKIT=1
+ : docker.io/labgrid/
+ : latest
+ main
+ local script_dir version
+ has_docker
+ command -v docker
+++ realpath ./build.sh
++ dirname /home/sgross/workspace/emlix/labgrid/dockerfiles/build.sh
+ script_dir=/home/sgross/workspace/emlix/labgrid/dockerfiles
++ python -m setuptools_scm
/usr/bin/python: No module named setuptools_scm
+ version=
++ get_docker_cmd ''
++ local docker_cmd
++ docker_cmd=
++ '[' -n '' ']'
++ has_docker
++ command -v docker
++ echo docker
+ docker_cmd=docker
+ cd /home/sgross/workspace/emlix/labgrid/dockerfiles/..
+ has_buildx docker
+ local docker_cmd
+ docker_cmd=docker
+ docker buildx --help
+ perform_docker_buildx_build docker /home/sgross/workspace/emlix/labgrid/dockerfiles ''
+ local docker_cmd script_dir version
+ docker_cmd=docker
+ script_dir=/home/sgross/workspace/emlix/labgrid/dockerfiles
+ version=
+ extra_args=("${@:4}")
+ for t in client exporter coordinator
+ docker buildx build --build-arg VERSION= --target labgrid-client -t docker.io/labgrid/client:latest -f /home/sgross/workspace/emlix/labgrid/dockerfiles/Dockerfile .
unknown flag: --build-arg
See 'docker --help'.
```

# Expected Behaviour

labgrid docker images should be build as stated in `dockerfiles/README.rst`

# System

## Linux
```
Linux phoebe 6.12.8-arch1-1 #1 SMP PREEMPT_DYNAMIC Thu, 02 Jan 2025 22:52:26 +0000 x86_64 GNU/Linux
```
## Distro
```
NAME="Arch Linux"
PRETTY_NAME="Arch Linux"
ID=arch
BUILD_ID=rolling
ANSI_COLOR="38;2;23;147;209"
HOME_URL="https://archlinux.org/"
DOCUMENTATION_URL="https://wiki.archlinux.org/"
SUPPORT_URL="https://bbs.archlinux.org/"
BUG_REPORT_URL="https://gitlab.archlinux.org/groups/archlinux/-/issues"
PRIVACY_POLICY_URL="https://terms.archlinux.org/docs/privacy-policy/"
LOGO=archlinux-logo
```
## Docker
```
Docker version 27.3.1, build ce1223035a
```

# Additional information

`buildx` is not available but `docker FOOBAR --help` will set `$?=0` due to the `--help` option
`docker buildx` will fix the false positive but might have undesired side effects if it exists or return a false negative.

Bastian-Krause, 2025-02-04:
Could you please test if #1597 fixes the issue?

sgross-emlix, 2025-02-04:
@Bastian-Krause LGTM
Cherrypicked your branch HEAD and
verified the both the negative case and the positive (installed `docker-buildx`).

However I did run into another error but this seems to be a "me-problem"
But for later reference

```
+ docker build --build-arg VERSION=24.0.3.dev56 --target labgrid-client -t docker.io/labgrid/client:latest -f /home/sgross/workspace/emlix/labgrid/dockerfiles/Dockerfile .
ERROR: BuildKit is enabled but the buildx component is missing or broken.
       Install the buildx component to build images with BuildKit:
       https://docs.docker.com/go/buildx/
```

PR #1597 (2025-02-04): dockerfiles/build.sh: fix buildx detection
**Description**
`docker <subcommand> --help` cannot be used to detect whether a sub command is available because it exits with 0 no matter if the sub command exists or not.

Use `docker buildx version` as a test for buildx support instead.

**Checklist**
- [ ] PR has been tested

Fixes: #1585