labgrid-project/labgrid #1622: BareboxDriver and (Small)UBootDriver: does not handle hex interrupt arguments (such as Ctrl-C) properly

pmelange, 2025-02-28:
In the BareboxDriver, UBootDriver, and SmallUBootDriver, the option sent to the bootloader to interrupt boot loading is done with some form of:
```
self.console.write(interrupstring.encode('ASCII'))
```

According to the documentation  
> use \\x03 for CTRL-C
or
> use “\x03” for CTRL-C

Neither of these work.

After using labgrid-client -vv I see the following is written to the console
```
CONSOLE SerialLogger.ma:   SerialDriver(main) > \x03␍␤
DEBUG   SerialDriver(ma:   Write 4 bytes: b'\\x03'
````
or
```
CONSOLE SerialLogger.ma:   SerialDriver(main) > \\x03␍␤
DEBUG   SerialDriver(ma:   Write 5 bytes: b'\\\\x03'
```

Neither of which is correct and causing the boot loader not to be interrupted.

Bastian-Krause, 2025-07-04:
Please share your actual yaml snippet.

I am pretty sure this works at our end:
```yaml
- BareboxDriver:
    interrupt: "\x03"              
    prompt: "barebox@[^:]+:[^ ]+ "
```

Bastian-Krause, 2025-07-04:
> According to the documentation
> 
> > use \x03 for CTRL-C
> > or
> > use “\x03” for CTRL-C

Where did you find these two versions?

pmelange, 2025-07-17:
> Where did you find these two versions?

https://github.com/labgrid-project/labgrid/blob/36955a4dc5603eabb5236d7edf95fdf9578b9798/doc/configuration.rst?plain=1#L1825

and

https://github.com/labgrid-project/labgrid/blob/36955a4dc5603eabb5236d7edf95fdf9578b9798/doc/configuration.rst?plain=1#L1909

pmelange, 2025-09-18:
I just remembered this. Sorry about the delay.

* example 1

configuration example taken from https://labgrid.readthedocs.io/en/latest/configuration.html#bareboxdriver

yaml file looks like this:
```
      boot_secret: '\x03'
```

debug console shows that 4 bytes are written:
```
CONSOLE SerialLogger.ma:     SerialDriver(main) > \x03␍␤
DEBUG   SerialDriver(ma:     Write 4 bytes: b'\\x03'
```

* example 2

configuration example taken from https://labgrid.readthedocs.io/en/latest/configuration.html#ubootdriver

yaml file looks like this:
```
    boot_secret: \\x03
```

debug console show that 5 bytes are written:
```
CONSOLE SerialLogger.ma:     SerialDriver(main) > \\x03␍␤
DEBUG   SerialDriver(ma:     Write 5 bytes: b'\\\\x03'
```

* proposed fix, but could probably be done cleaner

code change:
https://github.com/labgrid-project/labgrid/commit/315a968ae1f4fdbdcbbd89335ece4c30f55500fa

yaml file looking like this (from example 1)
```
      boot_secret: '\x03'
```

debug console shows that 1 byte is written
```
CONSOLE SerialLogger.ma:     SerialDriver(main) > ␍␤
DEBUG   SerialDriver(ma:     Write 1 bytes: bytearray(b'\x03')
```