labgrid-project/labgrid #1658: SSHDriver fails to launch commands in the background

adrian-martin, 2025-05-19:
Hi all,

I want to report an issue that I'm facing with the following Python code and the version v24.0.3 of Labgrid:
```Python
from labgrid.protocol.commandprotocol import CommandProtocol
from datetime import datetime

def execute_sleep(cmd_driver: CommandProtocol):
    print(f"Start: {datetime.now()}")
    cmd_driver.run("sleep 10 &")
    print(f"End: {datetime.now()}")
```

While the _ShellDriver_ works as expected when a command is launched in the background:
```Python
>>> shell_driver = target.get_driver('ShellDriver')
>>> target.activate(shell_driver)
>>> execute_sleep(shell_driver)
Start: 2025-05-19 18:33:46.764554
End: 2025-05-19 18:33:47.545368
>>> 
```

The _SSHDriver_ doesn't, always launching them in the foreground:
```Python
>>> ssh_driver = target.get_driver('SSHDriver')
>>> target.activate(ssh_driver)
>>> execute_sleep(ssh_driver)
Start: 2025-05-19 18:34:36.748644
End: 2025-05-19 18:34:46.782989
>>> 
```

This limitation can easily be bypassed by executing the blocking SSH command in a separate thread, but there is a consistency problem with the implementations of the _CommandProtocol_ regarding the commands attempted to be executed in the background.

Just for completion, I found [this pull request draft](https://github.com/labgrid-project/labgrid/pull/835) which seems to be working on something similar, but it looks more complex than what I wanted to report, so I decided to open the issue.

Best regards, Adrián

Bastian-Krause, 2025-06-03:
I'm not convinced that we should support background jobs (&). For the ShellDriver this results in unexpected output once the job is done which will confuse labgrid:
```
[1]+  Done                    sleep 10
```
Since that's local job control, it won't work for the SSHDriver.

`run()` and `run_check()` are meant to execute in a blocking fashion. As you already found, #835 is meant for everything else. As long as nobody works on that, I'd suggest using `systemd-run` as a work around. I have a helper class for that in a test suite. If you're interested, I could upstream that to https://github.com/labgrid-project/labgridhelper.

adrian-martin, 2025-06-27:
Hi Bastian,

Thanks for the `systemd-run` trick, I'll explore it. I didn't know about the *labgridhelper* repository, so I'll keep an eye on it too!

I hadn't considered that executing commands in the background introduces extra text in the console that might confuse Labgrid, but taking that into account, it makes sense to execute commands only in the foreground.

Best regards, Adrián