labgrid-project/labgrid #1660: Acquiring a place with a username without a "/" breaks labgrid-client

Troels51, 2025-05-22:
If i use the grpc protocol directly i can acquire a place on a username without a "/". This causes labgrid-client to break when you want to release it, as it tries to split the username into <hostname>/<username> multiple places. Which means you can never release the place again.

This is the sequence of grpc messages that can be done to get into this situation:
First create a [ClientStream](https://github.com/labgrid-project/labgrid/blob/6d8febb7263f772bb5d0b4aed85f268f7968412e/labgrid/remote/proto/labgrid-coordinator.proto#L6)
On that ClientStream you send a [StartupDone](https://github.com/labgrid-project/labgrid/blob/6d8febb7263f772bb5d0b4aed85f268f7968412e/labgrid/remote/proto/labgrid-coordinator.proto#L55) message with a name that does not contain a "/"
Then send a [AcquirePlaceRequest](https://github.com/labgrid-project/labgrid/blob/6d8febb7263f772bb5d0b4aed85f268f7968412e/labgrid/remote/proto/labgrid-coordinator.proto#L230)

Now the coordinator has a place that is acquired by a user without a "/"

guelfey, 2025-10-01:
We noticed the same underlying issue, but in a different way: we tried to put a value that contains slashes into `LG_USERNAME` when acquiring a place, which first is accepted, but then leads to problems later down the line in multiple components that assume exactly one slash in the owner name, e.g. a simple `labgrid-client who`:
```
Traceback (most recent call last):
  File "/nix/store/nmv02n65jq1rcw4arj1sa75d9g9m37l5-python3-3.12.10-env/lib/python3.12/site-packages/labgrid/remote/client.py", line 2146, in main
    args.func(session)
  File "/nix/store/nmv02n65jq1rcw4arj1sa75d9g9m37l5-python3-3.12.10-env/lib/python3.12/site-packages/labgrid/remote/client.py", line 400, in print_who
    host, user = place.acquired.split("/")
    ^^^^^^^^^^
ValueError: too many values to unpack (expected 2)
```

guelfey, 2025-10-02:
We indeed use LabGrid in a CI system. The username / hostname combination somewhat makes sense for the reservation, but if we want to see why a certain place is locked (and especially if we want to investigate why a place might still erroneously locked), we need to manually look through the webinterface of our CI system to find out a "job ID" that this user was executing at that time, in order to find out which CI job is actually locking the place. We instead tried to directly put the URL that leads to the job details in the username, which of course contains slashes. (As we realized this didn't work, we just put the job ID, which is not quite as comfortable as directly having the URL show up in `labgrid-client who`, but still pretty useful.)

I think it's valid to put restrictions on this value, but then it would be nice if these are enforced at "ingestion" already, instead of accepting the invalid values first but then leading to errors later on; or at least document the restrictions.

ozan956, 2026-11-12:
Hi, for CI use I would suggest using a username without slashes, for example ci-reponame-12345 with the job ID. The job ID is unique in most CI systems, so this should also work with many repositories. The runner name can go into LG_HOSTNAME, so labgrid-client who shows both. Its also the way we use at ADI, see: https://github.com/analogdevicesinc/hw-test/blob/main/hw_tests/labgrid.py#L61-L65

I agree with @guelfey that invalid values should be rejected early. Right now they are accepted first and only cause errors later, which makes the problem hard to understand.

I think the following needs to be done. The client should check the hostname and username and show a clear error if they contain a slash or other invalid characters. The coordinator should also check the name in StartupDone, so clients that use gRPC directly cannot create this situation. The limits for LG_USERNAME and LG_HOSTNAME should be documented as well.

 I can open a PR for this if it sounds good to you guys.