labgrid-project/labgrid #1661: Managedfile doesn't create a symlink for local tftp server

rvdgracht, 2025-06-02:
Hi,

When ManagedFile.sync_to_resource synces a file to a remote for use with tftp, a symlink to that file is created on the remote in the tftp server's internal directory.

When a local tftp server ('TFTPProvider' instead of 'RemoteTFTPProvider') is used, the file sync doesn't have to take place, but the symlink creation is also skipped, which was unexpected.

Bastian-Krause, 2025-06-03:
Yes, the symlinking should still happen. Maybe something like (untested) this in `labgrid/util/managedfile.py`:
```diff
-            if symlink is not None:
-                self.logger.info("Linking")
-                try:
-                    conn.run_check(f"test ! -e {symlink} -o -L {symlink}")
-                except ExecutionError:
-                    raise ManagedFileError(f"Path {symlink} exists but is not a symlink.")
-                # use short options to be compatible with busybox
-                # --symbolic --force --no-dereference
-                conn.run_check(f"ln -sfn {self.rpath}{os.path.basename(self.local_path)} {symlink}")
-
+        else:
+            conn = sshmanager.open("localhost")
+            self.rpath = os.path.dirname(self.local_path) + "/"
+
+        if symlink is not None:
+            self.logger.info("Linking")
+            try:
+                conn.run_check(f"test ! -e {symlink} -o -L {symlink}")
+            except ExecutionError:
+                raise ManagedFileError(f"Path {symlink} exists but is not a symlink.")
+            # use short options to be compatible with busybox
+            # --symbolic --force --no-dereference
+            conn.run_check(f"ln -sfn {self.rpath}{os.path.basename(self.local_path)} {symlink}")
```

If that works for you, please create a PR. Thanks!

Bastian-Krause, 2025-07-07:
Resolved via #1683.