labgrid-project/labgrid #1684: Feature request: Basic Windows support for the client

moqmar, 2025-07-07:
We'd like to control at least some resources through LabGrid from Windows (PowerShell) through the `-P` option (then using a Linux proxy) to make labgrid easy to use to acquire and at least somewhat control devices from a Windows machine.

Unfortunately, the labgrid-client doesn't support Windows at all - it would be nice to have at least some minimal functionality here. The absolute minimum we'd need is `places/resources/acquire/release/...` through an SSH proxy, as well as `power`.

Are there any plans/work to support Windows as a client already, or would a Pull Request on that topic have any chanve of being accepted?

jluebbe, 2025-07-07:
None of the maintainers use Windows (or MacOS for that matter), so we couldn't really maintain any non-Linux-support or answer user questions. We've heard that labgrid works fine in WSL2, might that be an option for you?

moqmar, 2026-10-12:
Just a quick update on this, as this came up at the LabGrid & Board Farming BOF at the Open Source Summit last week again: I felt motivated today to make the labgrid-client do something on Windows - meta stuff like acquiring & listing places works, and I have successfully gotten SSH to work (multiplexing needed to be disabled), which means that most of the stuff we actually need on Windows is working now.  
Of course many unix programs are missing on Windows, so many things will not work correctly out-of-the-box (telnet must be installed, snmpget doesn't really exist in the same way as on unix, etc.).

I have tried to keep the changes small for now and make them basically not touch anything on the Linux side, haven't run any tests on them yet though. The current state can be found at https://github.com/labgrid-project/labgrid/compare/master...ZEISS:labgrid:fix/windows-support. Adding some tests & a list of what exactly is working or not is probably a good goal for this issue before I'd create a PR.

Current progress:

- [x] Solve import errors for simple labgrid-client commands for basic functionality
- [x] Disable SSH multiplexing as that is not supported by Microsoft's Open SSH build
- [ ] Find a way to implement PtxExpect for windows (currently trying to use it raises a NotImplementedError)
- [ ] Figure out how bad not supporting non-blocking IO is (or what the Windows equivalent would be)
- [ ] Figure out what else is not working (testing!)