labgrid-project/labgrid #1749: disabling kernel messages (dmesg -n 1) is not realiable

pmelange, 2025-09-26:
After the ```login_prompt``` of "Please press Enter to activate this console.", the ShellDriver writes ```dmesg -n 1``` on the console to disable kernel messages on the console.  But this does not always work.  Sometimes the first couple characters are missing, sometime random characters are missing.  Here are some examples from the log...

```
CONSOLE SerialLogger.ma:     SerialDriver(main) > dmesg -n 1␤␍␤
INFO         StepLogger:     → SerialDriver.expect(pattern=['root@[-\\w()]+:[^ ]+ ', 'Please press Enter to activate this console.', 'Password: ', <class 'pexpect.exceptions.TIMEOUT'>]) 
CONSOLE SerialLogger.ma:      SerialDriver(main) < mesg -n 1␍␤
CONSOLE SerialLogger.ma:      SerialDriver(main) < -ash: mesg: not found␍␤
```

```
INFO         StepLogger:     → SerialDriver.expect(pattern=['root@[-\\w()]+:[^ ]+ ', 'Please press Enter to activate this console.', 'Password: ', <class 'pexpect.exceptions.TIMEOUT'>]) 
CONSOLE SerialLogger.ma:      SerialDriver(main) < ms- ␍␤
CONSOLE SerialLogger.ma:      SerialDriver(main) < -ash: ms-: not found␍␤
```

Perhaps some kind of check to see if ```dmesg -n 1``` worked would be useful, which can be done by checking the contents of ```/proc/sys/kernel/printk```.  Or possibly it would be beneficial to add a sleep (user configurable with a default of 0 seconds) before running ```dmesg -n 1```.

pmelange, 2025-09-28:
note: the expectation of ```\w+: not found``` works for sh and ash.  An expectation of ```\w+: command not found``` would be needed for bash and zsh.  I haven't tried other shells.

ynezz, 2025-09-28:
>  Sometimes the first couple characters are missing, sometime random characters are missing.

FYI https://github.com/labgrid-project/labgrid/pull/945